RMerl/asuswrt-merlin.ng #631: I'd like to add one line

odkrys, 2020-11-05:
```
include ../../.config
```
The code above is omitted in the src-rt-5.02axhnd.675x sdk and this creates an empty -D flag when compiling in entware.

https://github.com/RMerl/asuswrt-merlin.ng/blob/2b31617cd03c4fe9bb75fdeb2ca06da1f91c3f65/release/src-rt-5.02axhnd/kernel/linux-4.1/Makefile#L420

https://github.com/RMerl/asuswrt-merlin.ng/blob/2b31617cd03c4fe9bb75fdeb2ca06da1f91c3f65/release/src-rt-5.02axhnd.675x/kernel/linux-4.1/Makefile#L417

```
-Wno-format-security -std=gnu89 -D -mabi=aapcs-linux -mno-thumb-interwork 
```

RMerl, 2020-11-06:
I'm still cleaning up the 675x SDK from 40577, I'll take a look once I'm done with it.

RMerl, 2020-11-18:
Can't do that, this creates an infinite loop.

```
make[6]: Entering directory '/home/merlin/amng.ax58/release/src-rt-5.02axhnd.675x/kernel/linux-4.1'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
In file included from scripts/kconfig/zconf.tab.c:2534:
scripts/kconfig/confdata.c: In function ‘conf_write’:
scripts/kconfig/confdata.c:776:20: warning: ‘.tmpconfig.’ directive writing 11 bytes into a region of size between 1 and 4097 [-Wformat-overflow=]
  776 |   sprintf(tmpname, "%s.tmpconfig.%d", dirname, (int)getpid());
      |                    ^~~~~~~~~~~~~~~~~
scripts/kconfig/confdata.c:776:3: note: ‘sprintf’ output between 13 and 4119 bytes into a destination of size 4097
  776 |   sprintf(tmpname, "%s.tmpconfig.%d", dirname, (int)getpid());
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  HOSTLD  scripts/kconfig/conf
Using .config as base
Merging ./kernel/configs/../../.config
#
# merged configuration written to ./.config (needs make)
#
Using .config as base
Merging ./kernel/configs/../../.config
#
# merged configuration written to ./.config (needs make)
#
Using .config as base
Merging ./kernel/configs/../../.config
#
# merged configuration written to ./.config (needs make)
#
Using .config as base
Merging ./kernel/configs/../../.config
#
# merged configuration written to ./.config (needs make)
#
Using .config as base
Merging ./kernel/configs/../../.config
```

closed 2020-11-18 by RMerl