RMerl/asuswrt-merlin.ng #632: Null Outbound Trend Micro Requests/Packets

TensorTom, 2020-11-05:
Many useful features require the user to accept Trend Micro's [EULA](https://www.asus.com/Microsite/networks/Trend_Micro_EULA/), [Privacy Policy](https://www.trendmicro.com/en_us/about/legal/privacy.html), and [Data Collection Notice](https://helpcenter.trendmicro.com/en-us/article/TMKA-20275). For the entire life of my device, I have left these features disabled. Not because I don't want to use them but because I don't telemetry and who knows what else sent to Trend Micro.

Trend Micro's practices and motivations [are less than transparent](https://malwaretips.com/threads/why-does-trend-micro-send-telemetry-to-data-mining-firms.71932/) and it would be super-useful to be able to enable an option that will still allow downloading generic rule files and whatnot from Trend Micro while at the same time blocking any egress traffic to Trend Micro's servers.

TensorTom, 2020-11-05:
Shouldn't this be possible with user scripts/entware? I know it could be done simply by some hosts entries but I prefer something a bit less ham-handed.

RMerl, 2020-11-05:
You are asking me to make my firmware capable of breaching the EULA.  That`s a definiitve no on that one, even assuming it were technically possible (which it isn't).

closed 2020-11-05 by RMerl