RMerl/asuswrt-merlin.ng #637: Let's Encrypt Certificate for HTTPS access auto-renewal fails

DavidDosoudil, 2020-11-16:
Hi,

I've noticed my auto generated certificate was expired and it didn't renew automatically. I have tried to renew it manually (on DDNS page) and it always ends up with an error:
```
Nov 16 15:30:11 kernel: [Mon Nov 16 15:30:11 GMT 2020] Renew: 'domain.xxx'
Nov 16 15:30:15 kernel: [Mon Nov 16 15:30:15 GMT 2020] Standalone mode.
Nov 16 15:30:16 kernel: [Mon Nov 16 15:30:16 GMT 2020] Registering account
Nov 16 15:30:19 kernel: [Mon Nov 16 15:30:19 GMT 2020] Already registered
Nov 16 15:30:19 kernel: [Mon Nov 16 15:30:19 GMT 2020] ACCOUNT_THUMBPRINT='xxxxxx'
Nov 16 15:30:19 kernel: [Mon Nov 16 15:30:19 GMT 2020] Single domain='domain.xxx'
Nov 16 15:30:20 kernel: [Mon Nov 16 15:30:20 GMT 2020] Getting domain auth token for each domain
Nov 16 15:30:23 kernel: [Mon Nov 16 15:30:23 GMT 2020] Getting webroot for domain='domain.xxx'
Nov 16 15:30:23 kernel: [Mon Nov 16 15:30:23 GMT 2020] Verifying: domain.xxx
Nov 16 15:30:23 kernel: [Mon Nov 16 15:30:23 GMT 2020] Standalone mode server
Nov 16 15:30:29 kernel: [Mon Nov 16 15:30:29 GMT 2020] domain.xxx:Verify error:Fetching http://domain.xxx/.well-known/acme-challenge/xxxxxxxx: Connection refused
Nov 16 15:30:30 kernel: [Mon Nov 16 15:30:30 GMT 2020] Please add '--debug' or '--log' to check more details.
Nov 16 15:30:30 kernel: [Mon Nov 16 15:30:30 GMT 2020] See: https://github.com/Neilpang/acme.sh/wiki/How-to-debug-acme.sh
```
The domain is registered and correctly resolved, if I enable WAN webadmin access, it works (albeit with certificate warning).

My guess is, that Let's Encrypt needs **BOTH** ports 80 and 443 to do the ACME file challenge and since only 443 could be enabled for WAN, it fails to access the file and thus fails to renew the certificate. I haven't tried it yet, but I suppose any shell attempt to do the same will fail for the same reason.

Is there a way how to enable this to work? For example opening 80/443 for a short duration just for certificate renewal and than back to normal? Or is there any other way how to generate or renew this SSL certificate not using the challenge method?

Thanks

RMerl, 2020-11-17:
The Let's Encrypt code is closed source, and outside of my control.  While it uses acme.sh, the code handling its calls is closed source.

closed 2020-11-17 by RMerl

phantomski77, 2020-11-21:
No problem @RMerl , I understand it must be frustrating ASUS making stuff increasingly closed source. It should be exactly opposite for network security.
Anyway, thanks for your reply, here’s my very basic interim solution:
[https://www.snbforums.com/threads/solution-lets-encrypt-ddns-wan-admin-domain-certificate-expired-not-auto-renewed.67762/](url)

jaklan, 2020-12-23:
@phantomski77 it seems that permanent location is `/jffs/cert.tgz`, which includes `etc` folder with `cert.pem` and `key.pem` inside