RMerl/asuswrt-merlin.ng #646: Not importing OpenVPN certs

andrew0007, 2020-11-26:
Hello, when I am trying to add a new OpenVPN Client config, but I get the following error:
```
Options error: You must define CA file (--ca) or CA path (--capath)
Starting OpenVPN failed...
```

It seems like the .ovpn file is not properly saved.
I am using RT-AC86U_384.19_0

What can be the issue? Thanks.

CloudStu, 2020-11-28:
Same for me, using RT-AX86U_386.1_alpha3-g2f4644f6a8

My .ovpn file has both an embedded 

```
<ca>
-----BEGIN CERTIFICATE-----
```

and an embedded 

```
<tls-auth>
#
# 2048 bit OpenVPN static key
#
-----BEGIN OpenVPN Static key V1-----
```

When importing the .ovpn, whilst the other settings are populated, these blocks do not appear in the keys and certificates pop-up. If I then add them manually, they don't stick after saving the page. When trying to turn on the client, I see the same error as the original poster in the system logs.

```
Nov 28 01:11:09 rc_service: httpd 1300:notify_rc restart_vpnclient1
Nov 28 01:11:09 ovpn-client1[21439]: Options error: You must define CA file (--ca) or CA path (--capath)
Nov 28 01:11:09 ovpn-client1[21439]: Use --help for more information.
Nov 28 01:11:09 openvpn: Starting OpenVPN client 1 failed!
```

DidierACHE, 2020-12-16:
Experiencing the exact same issue than CloudStu using 386.1_beta2 on an RT-AX86U

skaman, 2020-12-27:
I had the same problem. The webif wasn't saving my certificates (even with both automatic import and manual editing). At the end I tried to edit my certificates with Firefox (instead of Edge) and it worked.

RMerl, 2020-12-28:
@andrew0007
> What can be the issue? Thanks.

Works for me.  Please provide the content of your ovpn file.

@CloudStu
> Same for me, using RT-AX86U_386.1_alpha3-g2f4644f6a8

Please try again with the latest beta build.  Also try removing the comment at the top of your static key (i.e. remove the three lines starting with # )

@DidierACHE, @skaman

Need more info, as not everyone's case is the same, especially since I cannot reproduce the issue myself.

andrew0007, 2020-12-29:
@RMerl 
I tried to import the ovpn file from NordVpn.
For example this one: https://downloads.nordcdn.com/configs/files/ovpn_udp/servers/fr595.nordvpn.com.udp.ovpn

DidierACHE, 2020-12-29:
@RMerl 
It does work for me now on Beta 3 ( Using NordVpn) Thanks! 
Happy Holidays everyone!!!

RMerl, 2020-12-29:
I do all of my VPN client testing with NordVPN myself, so it's the most tested scenario, and is known to work.

Make sure your JFFS partition didn't get filled up.  There should be a warning when that is the case.

andrew0007, 2021-01-02:
Hi, I confirm that it is working properly by doing "Format JFFS partition at next boot" under Administration -> System after the installation of the Merlin Firmware. Tested with RT-AC86U_384.19_0.
Thanks a lot.

closed 2021-01-03 by RMerl