RMerl/asuswrt-merlin.ng #647: IPSet kernel modules missing for AX86U

Adamm00, 2020-11-27:
I know this model is still in the alpha stages of adding support, but just incase this small detail was overlooked;

> skynet@RT-AX86U-38B8:/# lsmod
> Module                  Size  Used by    Tainted: P
> tdts_udbfw             45602  1
> tdts_udb              249264  1 tdts_udbfw
> tdts                  453179  2 tdts_udbfw,tdts_udb
> sr_mod                 14635  0
> cdrom                  29513  1 sr_mod
> thfsplus              102277  0
> tntfs                 471824  0
> tfat                  280872  0
> uas                    15645  0
> usb_storage            54203  2 uas
> sg                     28806  0
> sd_mod                 31178  2
> scsi_mod              171485  5 sr_mod,uas,usb_storage,sg,sd_mod
> dhd                   975442  0
> wl                   6345625  0
> nf_nat_sip              9604  0
> nf_conntrack_sip       24344  1 nf_nat_sip
> nf_nat_h323             6991  0
> nf_conntrack_h323      41585  1 nf_nat_h323
> nf_nat_rtsp             4528  0
> nf_conntrack_rtsp       7915  1 nf_nat_rtsp
> nf_nat_ftp              2099  0
> nf_conntrack_ftp        7469  1 nf_nat_ftp
> xt_length               1245  2
> cdc_mbim                5332  0
> qmi_wwan               14529  0
> cdc_wdm                10545  2 cdc_mbim,qmi_wwan
> cdc_ncm                16787  1 cdc_mbim
> rndis_host              6121  0
> cdc_ether               4896  1 rndis_host
> ax88179_178a           16204  0
> asix                   22797  0
> libphy                 27753  1 asix
> cdc_acm                18547  0
> usbnet                 20580  7 cdc_mbim,qmi_wwan,cdc_ncm,rndis_host,cdc_ether,ax88179_178a,asix
> mii                     4341  3 ax88179_178a,asix,usbnet
> ohci_pci                3260  0
> ohci_platform           5283  0
> ohci_hcd               31852  2 ohci_pci,ohci_platform
> ehci_pci                4010  0
> ehci_platform           5831  0
> ehci_hcd               40022  2 ehci_pci,ehci_platform
> xhci_pci                4006  0
> xhci_plat_hcd           4622  0
> xhci_hcd              103270  2 xhci_pci,xhci_plat_hcd
> bcm_usb                 1590  0
> usbcore               166604 21 uas,usb_storage,cdc_mbim,qmi_wwan,cdc_wdm,cdc_ncm,rndis_host,cdc_ether,ax88179_178a,asix,cdc_acm,usbnet,ohci_pci,ohci_platform,ohci_hcd,ehci_pci,ehci_platform,ehci_hcd,xhci_pci,xhci_plat_hcd,xhci_hcd
> usb_common              2813  1 usbcore
> dpsta                  23014  2 dhd,wl
> cfg80211              224695  2 dhd,wl
> igs                    24676  2 dhd,wl
> emf                    17150  3 dhd,wl,igs
> hnd                   315935  5 dhd,wl,dpsta,igs,emf
> otp                     2777  0
> bcm_thermal             6541  0
> bcmspu                 66743  0
> bcmpdc                 10168  1 bcmspu
> pwrmngtd                3501  0
> wfd                    32344  2 dhd,wl
> bcm_pcie_hcd           30395  0
> bcmmcast               66292  5 dhd,igs,emf,wfd
> nciTMSkmod            381117  0
> nciexLinuxETH          16973  1 nciTMSkmod
> nciLservices            6968  1 nciTMSkmod
> pktrunner              56323  0
> bcm_enet              205325  1 tdts_udbfw
> cmdlist                78481  1 pktrunner
> pktflow               266779  2 tdts_udbfw,pktrunner
> rdpa_cmd              108830  0
> bcm_ingqos            221124  1 rdpa_cmd
> chipinfo                1769  0
> rdpa_mw                30403  1 rdpa_cmd
> rdpa_usr               34303  0
> rdpa                 1736569  2 wfd,bcm_enet
> rdpa_gpl_ext            1377  1 rdpa_cmd
> rdpa_gpl               28007  8 dhd,wfd,pktrunner,bcm_enet,rdpa_cmd,rdpa_mw,rdpa_usr,rdpa
> bcmvlan               111895  0
> bdmf                 1274123  9 dhd,wfd,pktrunner,bcm_enet,rdpa_cmd,rdpa_mw,rdpa_usr,rdpa,rdpa_gpl
> rdp_fpm                31982  1 rdpa
> bcmlibs                22223  7 dhd,wl,wfd,pktrunner,bcm_enet,pktflow,rdpa
> wlcsm                  18061 73 dhd,hnd,bcm_pcie_hcd

> skynet@RT-AX86U-38B8:/# ls /lib/modules/"$(uname -r)"/kernel/net/netfilter
> nf_conntrack_ftp.ko   nf_conntrack_rtsp.ko  nf_conntrack_tftp.ko  nf_nat_ftp.ko         nf_nat_tftp.ko        xt_TEE.ko             xt_blog.ko            xt_condition.ko       xt_geoip.ko           xt_hl.ko              xt_quota.ko
> nf_conntrack_h323.ko  nf_conntrack_sip.ko   nf_dyndscp.ko         nf_nat_sip.ko         xt_HL.ko              xt_TPROXY.ko          xt_comment.ko         xt_flowlabel.ko       xt_hashlimit.ko       xt_length.ko          xt_socket.ko

RMerl, 2020-11-27:
Which modules specifically are missing?  I did apply the same config_base.6a changes to that kernel that I did to the other 4.1.xx kernels.

Adamm00, 2020-11-27:
> Which modules specifically are missing? I did apply the same config_base.6a changes to that kernel that I did to the other 4.1.xx kernels.

There should be a folder "/lib/modules/"$(uname -r)"/kernel/net/netfilter/ipset" containing them all which is missing entirely. I specifically test for "ip_set_hash_ipmac.ko" but I'll fire up my AX88U and find out the full list.

Adamm00, 2020-11-27:
> skynet@RT-AX88U-DC28:/lib/modules/4.1.51/kernel/net/netfilter/ipset# ls
> ip_set.ko                  ip_set_bitmap_ipmac.ko     ip_set_hash_ip.ko          ip_set_hash_ipmark.ko      ip_set_hash_ipportip.ko    ip_set_hash_mac.ko         ip_set_hash_netiface.ko    ip_set_hash_netport.ko     ip_set_list_set.ko
> ip_set_bitmap_ip.ko        ip_set_bitmap_port.ko      ip_set_hash_ipmac.ko       ip_set_hash_ipport.ko      ip_set_hash_ipportnet.ko   ip_set_hash_net.ko         ip_set_hash_netnet.ko      ip_set_hash_netportnet.ko

RMerl, 2020-11-27:
Seems like Asus has them set as built-in rather than external modules:

```
admin@RT-AX86U-5F38:/lib/modules/4.1.52/kernel/net# zcat /proc/config.gz | grep IP_SET
CONFIG_IP_SET=y
CONFIG_IP_SET_MAX=256
CONFIG_IP_SET_BITMAP_IP=y
CONFIG_IP_SET_BITMAP_IPMAC=y
CONFIG_IP_SET_BITMAP_PORT=y
CONFIG_IP_SET_HASH_IP=y
CONFIG_IP_SET_HASH_IPMARK=y
CONFIG_IP_SET_HASH_IPPORT=y
CONFIG_IP_SET_HASH_IPPORTIP=y
CONFIG_IP_SET_HASH_IPPORTNET=y
CONFIG_IP_SET_HASH_MAC=y
CONFIG_IP_SET_HASH_NETPORTNET=y
CONFIG_IP_SET_HASH_NET=y
CONFIG_IP_SET_HASH_NETNET=y
CONFIG_IP_SET_HASH_NETPORT=y
CONFIG_IP_SET_HASH_NETIFACE=y
CONFIG_IP_SET_LIST_SET=y
```

Adamm00, 2020-11-27:
> Seems like Asus has them set as built-in rather than external modules:
> 
> ```
> admin@RT-AX86U-5F38:/lib/modules/4.1.52/kernel/net# zcat /proc/config.gz | grep IP_SET
> CONFIG_IP_SET=y
> CONFIG_IP_SET_MAX=256
> CONFIG_IP_SET_BITMAP_IP=y
> CONFIG_IP_SET_BITMAP_IPMAC=y
> CONFIG_IP_SET_BITMAP_PORT=y
> CONFIG_IP_SET_HASH_IP=y
> CONFIG_IP_SET_HASH_IPMARK=y
> CONFIG_IP_SET_HASH_IPPORT=y
> CONFIG_IP_SET_HASH_IPPORTIP=y
> CONFIG_IP_SET_HASH_IPPORTNET=y
> CONFIG_IP_SET_HASH_MAC=y
> CONFIG_IP_SET_HASH_NETPORTNET=y
> CONFIG_IP_SET_HASH_NET=y
> CONFIG_IP_SET_HASH_NETNET=y
> CONFIG_IP_SET_HASH_NETPORT=y
> CONFIG_IP_SET_HASH_NETIFACE=y
> CONFIG_IP_SET_LIST_SET=y
> ```
> 
> They actually enable ipset because it's used by their TCPLUGIN that's currently specific to that model (Tencent-related gaming accelerator).

Ah okay interesting. Another issue was that using IPSet was causing me to kernel panic during Skynet’s startup process. I will have to investigate further and report back as to what specifically triggers it as at the time I just assumed it was to do with the modules “missing”.

RMerl, 2020-11-28:
Can you provide a narrower test case?  The crash seem to be related to adding an ipv4 hash, possibly a NULL call to kfree(), but that doesn't tell me which actual command triggered it.

```
May  5 15:05:11 crashlog: <0>Call trace:
May  5 15:05:11 crashlog: <4>[<ffffffc000133bfc>] kfree+0x74/0xf8
May  5 15:05:11 crashlog: <4>[<ffffffc00047c0a0>] hash_net4_add+0x410/0x470
May  5 15:05:11 crashlog: <4>[<ffffffc00047a4f0>] hash_net4_uadt+0x308/0x398
May  5 15:05:11 crashlog: <4>[<ffffffc00045eac8>] call_ad+0x98/0x1c0
May  5 15:05:11 crashlog: <4>[<ffffffc00045f050>] ip_set_uadd+0x208/0x268
May  5 15:05:11 crashlog: <4>[<ffffffc000441cdc>] nfnetlink_rcv_msg+0x254/0x268
May  5 15:05:11 crashlog: <4>[<ffffffc00043d93c>] netlink_rcv_skb+0xc4/0xf8
May  5 15:05:11 crashlog: <4>[<ffffffc000441fb0>] nfnetlink_rcv+0x240/0x460
May  5 15:05:11 crashlog: <4>[<ffffffc00043d218>] netlink_unicast+0x150/0x228
May  5 15:05:11 crashlog: <4>[<ffffffc00043d6e8>] netlink_sendmsg+0x2f8/0x358
May  5 15:05:11 crashlog: <4>[<ffffffc0003d9d88>] sock_sendmsg+0x18/0x58
May  5 15:05:11 crashlog: <4>[<ffffffc0003dcd98>] SyS_sendto+0xa8/0x128
May  5 15:05:11 crashlog: <0>Code: f8626800 37780440 f9401820 d538d094 (f9400013) 
May  5 15:05:11 crashlog: <4>---[ end trace e973731c5944cedb ]---
```
> Also fwiw, CONFIG_IP_SET_HASH_IPMAC / ip_set_hash_ipmac.ko isn't enabled whereas it is on all other models.

That's odd because it's definitely set in config_base.6a, as a module.  Could be that this module is not part of the 4.1.xx kernel (and I couldn't apply the previous kernel version patches as they don't apply cleanly to 4.1.52), so the config_base config line doesn't do anything.

RMerl, 2020-11-28:
For some reason the ipset 6.32 modules fail to build if set as built-in instead of as modules.  I was able to apply and build with the 6.32 patches after making them all build as modules.

Give this test image a try to see if those 6.32 patches resolved your issues.

https://1drv.ms/u/s!AuCcWdNeYuXMgZhntxaOzmeFKk5LbQ?e=aeqH4N

Adamm00, 2020-11-28:
> For some reason the ipset 6.32 modules fail to build if set as built-in instead of as modules. I was able to apply and build with the 6.32 patches after making them all build as modules.
>
> Give this [...]

Thanks seems to have resolved the issue, tried to stress test it and couldn't get it to crash again.

The only thing left would be ipset complaining about userspace and kernel supporting different versions but I'm not sure if there is any downside besides aesthetically.

> skynet@RT-AX86U-38B8:/lib/modules/4.1.52/kernel/net/netfilter/ipset# ipset -v
> ipset v7.6, protocol version: 7
> Warning: Kernel support protocol versions 6-6 while userspace supports protocol versions 6-7

RMerl, 2020-11-28:
> Thanks seems to have resolved the issue, tried to stress test it and couldn't get it to crash again.

Was fixed either by 6.32 or by the post-6.32 fix that was included.

> The only thing left would be ipset complaining about userspace and kernel supporting different versions but I'm not sure if there is any downside besides aesthetically.

Version 7.xx of userspace tool support both protocol v6 and protocol v7.  The kernel is on 6.32, therefore it uses protocol v6.  In itself there is nothing wrong with it, I decided to upgrade to userspace version 7.6 so to be in sync with Asus.

closed 2020-11-29 by Adamm00