RMerl/asuswrt-merlin.ng #652: Request update 5 GHz channels in CN

NakornC, 2020-11-29:
/router/www/general.js

Please update the 5 GHz channels in China if you are available, thank you.
(149, 153, 157, 161) -> (36, 40, 44, 48, 64, 149, 153, 157, 161, 165)

https://www.lairdconnect.com/support/faqs/what-channels-are-supported-both-24-ghz-and-5ghz-band-most-countries

paldier, 2020-11-30:
You should do that in wifi driver and it's closed code. 
for example:[https://github.com/MerlinRdev/ac85p-merlin/commit/7eb5d7efb1d735cdbd991a310fef4ea9d85634f0/release/src-ra-5010/linux/linux-3.10.14.x/drivers/net/wireless/ralink/mt7615/embedded/common/rt_channel.c](https://github.com/MerlinRdev/ac85p-merlin/commit/7eb5d7efb1d735cdbd991a310fef4ea9d85634f0/release/src-ra-5010/linux/linux-3.10.14.x/drivers/net/wireless/ralink/mt7615/embedded/common/rt_channel.c)
brcm:wlc_clm_data.c

RMerl, 2020-12-28:
Everything related to wifi is closed source and outside of my control.

closed 2020-12-28 by RMerl