RMerl/asuswrt-merlin.ng #653: Update nano to version 5.4

feranick, 2020-12-04:
Changelogs from version 5.2 in master:

v.54
Moving the cursor now skips over combining characters (another zero-width characters). Deleting a character deletes also any succeeding zero-width characters, but backspacing deletes just one character at a time.

v.5.3
Option 'set stateflags' makes nano show the state of auto-indenting, the mark, hard-wrapping, macro recording, and soft-wrapping in the title bar. The flags take the place of "Modified", and a modified buffer is instead indicated by an asterisk (*) after its name.
Nano no longer by default tries using libmagic to determine the type of a file (when neither filename nor first line gave a clue), because in most cases it is a waste of time. It requires using the option --magic or -! or 'set magic' to make nano try libmagic.
The color of the indicator can be changed with 'set scrollercolor'.
--

JackMerlin, 2020-12-30:
I think you will get more help if you make a feature request on the SNB forum. Like Rmerlin said "_The issue tracker is for development/bug tracking purposes only!_"

feranick, 2020-12-30:
Filing feature requests about nano worked in the past and I find it a cleaner way to track it than submit to a forum.

JackMerlin, 2020-12-30:
> Filing feature requests about nano worked in the past and I find it a cleaner way to track it than submit to a forum.

Don't do this, you are wasting other people's time. Any request here has the highest priority, because they should be serious errors that can be reproduced, not feature requests!

feranick, 2020-12-30:
Chill buddy. Look at the title: Update nano to version xx. you know what you get. Merlin responded to questions before about similar requests, and was very helpful at that. 

Because this rubs people the wrong way, I'll close the bug and go on with my life. I don't have myself time to waste.

closed 2020-12-30 by feranick