RMerl/asuswrt-merlin.ng #656: wget/TOR need to be "autoreconf'd" for Ubuntu 20.04

Adamm00, 2020-12-10:
Due to Ubuntu 20.04 having a newer version of aclocal (1.16.1) compiling fails for both wget and TOR as they request aclocal (1.15). This can be fixed via manually running the following command in each directory prior to building;

"$HOME/$FWPATH/../src/router/wget"
"$HOME/$FWPATH/../src/router/tor"

`autoreconf -f -i`

Not sure if there is a way to correct this so it will compile on both versions without additional steps required as this is the only change needed for simultaneous Ubuntu 18.04/20.04 support.

https://github.com/Adamm00/amcfwm/commit/625a62fd942c074ae157f44ad3eac7f3bdb4b5eb

RMerl, 2020-12-10:
I use 20.04 here and don`t have any issue, both on my main dev VM and on the WSL2 environment I tested a few months ago.  See the note about 20.04 posted on the https://github.com/RMerl/asuswrt-merlin.ng/wiki/Compiling-under-WSL2 guide:

```
sudo apt install automake-1.15
```

Adamm00, 2020-12-10:
My mistake, was the first step I tried but didn't type the specific version number. Thanks

closed 2020-12-10 by Adamm00