RMerl/asuswrt-merlin.ng #657: Enable caching on JavaScript files

trajano, 2020-12-14:
Caching is enabled for /js/jquery.js but I can it also be done on 

/help.js
/validator.js
/state.js
/popup.js
/httpApi.js
/general.js

Trying to find out where in the code it is explicitly setting the `no-cache` on this one but there's quite a few matches on `no-cache` specifically on the asp pages which is expected.

Also when enabling the cache, it should have an ETag which could be the firmware version so if the firmware gets updated the cache is also invalidated.

RMerl, 2020-12-14:
No, it cannot.  These contain dynamic content that need to be injected by the web server on each access.

trajano, 2020-12-14:
Ah I see where it's being done

https://github.com/RMerl/asuswrt-merlin.ng/blob/master/release/src/router/www/state.js#L246

I am wondering if we can somehow inject a file that contains all the nvram variables and the path functions and store it into a `window` level variable then make the existing js pages access it.

Although even that may break things if they have some other computed value.

trajano, 2020-12-14:
We can probably do `/require/require.min.js` since that's minified.  and also does not have `<%` maybe a more specific rule is anything with `.min.js` should be cached.

RMerl, 2020-12-28:
Caching these two would be doable, as I see they pretty much never changes.  I generally try to be conservative with caching, since 90% of webui access is done over a LAN, therefore you aren't gaining much out of browser-side caching, unlike a public website that has to deal with much higher latency.

Note that jquery-ui.js is only used AFAIK by the parental control pages, so it's very rarely accessed.  Caching is mostly useful here for content that gets used on multiple pages, as the cache time is kept fairly low - just enough to cache throughout one session.

I'm also very conservative with upgrading any of the bundled Javascripts, as the vast majority of the code that make use of it is written by Asus, therefore things may break at any moment if I use different versions of core libraries than those they use for their own development.  They also occasionally make changes to them, like they did with Chart.js, which forced me to use a separate copy of Chart.js for my own use.

trajano, 2020-12-29:
At least require.min should be cached.  That seems to be the common one.  I think it's because I may have an older router, but the HTTP server appears to be single threaded so it is sending things one at a time.

RMerl, 2020-12-31:
Adding these two should be fine as they very rarely change.

trajano, 2020-12-31:
Can you point out which part of the code does the caching choice for jquery.js?  I can't find it when I did this

https://github.com/search?q=jquery.js+repo%3ARMerl%2Fasuswrt-merlin.ng+language%3AC&type=Code&ref=advsearch&l=C&l=

trajano, 2020-12-31:
When doing the change you should have an `ETag` also set as part of the response so that it can be checked in case there's a new version of the firmware and upstream changed their files (but that can probably come later)

RMerl, 2020-12-31:
> Can you point out which part of the code does the caching choice for jquery.js? I can't find it when I did this
> 
> https://github.com/search?q=jquery.js+repo%3ARMerl%2Fasuswrt-merlin.ng+language%3AC&type=Code&ref=advsearch&l=C&l=

https://github.com/RMerl/asuswrt-merlin.ng/blob/master/release/src/router/httpd/web.c#L19422

> When doing the change you should have an `ETag` also set as part of the response so that it can be checked in case there's a new version of the firmware and upstream changed their files (but that can probably come later)

That would require more work, and I don't want to spend too much time overengineering things for something that will have a very minimal impact on the router's general performance.  Adding these two to caching only involved two lines of code, so it was a quick enough change.

trajano, 2021-01-01:
@RMerl agreed, hopefully it will be in the next beta so I can see it.  Worst case scenario the users can simply clear the browser cache if the firmware was updated.

trajano, 2021-01-03:
Closed since it's already in master.

closed 2021-01-03 by trajano