RMerl/asuswrt-merlin.ng #659: Create a Dockerfile to build the image

trajano, 2020-12-14:
I am thinking of taking https://github.com/RMerl/asuswrt-merlin.ng/wiki/Compiling-under-WSL2 and somehow convert it to use Docker perhaps a multistage build if needed to assemble the firmware images.  At least the process should be more "repeatable"

RMerl, 2020-12-28:
Feel free to do so.  I know that the DSL form developed by Gnuton makes use of CI to generate his own buiilds, so it's already out there.

Personally I have no plan to switch to a docker-based build system.  I prefer to build on my own machine, which is way faster than any cloud-based solution (outside if crazy expensive solutions that provide multiple CPU cores).

closed 2020-12-28 by RMerl

trajano, 2020-12-29:
I don't expect you to do a docker-based build for yourself, it is more to allow external contributors to be able to simplify their development setup rather than creating a whole new VM.