RMerl/asuswrt-merlin.ng #660: Cannot edit Administration - System settings without changing login name

mpratt14, 2020-12-17:
AC68P
384.19

I am trying to enable Reboot Scheduler, but Apply is being blocked by Router Login Name, but it is left unchanged...
the username is only numbers and has not been changed for many updates...probably since 2017.

the message is
`Only alphanumeric characters and dash symbol are accepted. The first character cannot be a dash - .`

mpratt14, 2020-12-17:
I would recommend that making Router Login Name blank is accepted as "no change" instead of also erroring with `Username cannot be blank.`

However, I don't know if this is Asus code or something you control

hurda, 2020-12-17:
>the username is only numbers and has not been changed for many updates...probably since 2017.
One of the recent updates added a username-validator, which requires the name to start with a letter, not with a number: https://github.com/RMerl/asuswrt-merlin.ng/blob/c514f9f0b3f279dc1ea91ede4c92ab3e02beb8fa/release/src/router/www/validator.js#L372

mpratt14, 2020-12-17:
so the problem is the regex in the `account_name` function
and it is returning the messages `JS_valid_account_name` and `JS_valid_account_name_first_char`

and for username being blank, it is in the file
release/src/router/lighttpd-1.4.39/external_file/js_src/setting.js

in the `OK` function it checks many things
and returns the message `msg_err_account_blank`

is it a mistake that the regex makes it start with a letter? or there is some reason for this?

mpratt14, 2020-12-17:
I believe true alphanumeric with dash, but not starting with dash would be

` 	var re = new RegExp(/^[a-zA-Z0-9]+[-a-zA-Z0-9]*$/i); `

if this is not desired for some reason, the error message `JS_valid_account_name` should be edited

thanks in advance, I love the project 👍🏽

hurda, 2020-12-17:
You're not the first to run into this:
#557

https://www.snbforums.com/threads/bug-administration-system-router-login-name-errors-384-18-19-at-least.65640/

https://www.snbforums.com/threads/asuswrt-merlin-384-19-login-name-cant-start-with-a-number.67021/

mpratt14, 2020-12-17:
all these changes were added by a GPL commit
cdfc71fd209c30d3cb0977ecedf71744d3492d8c

I can see that since this is GPL it might not be changed, but if it is I guess a better error message is enough

RMerl, 2020-12-28:
Asus might have reasons to reject the use of a username that begins with a number, therefore I have no plan on changing this behaviour, as I can't tell what else it might break.

closed 2020-12-28 by RMerl

hurda, 2020-12-28:
I think the problem isn't the username not being valid, but the username becoming not valid to be used for changing some settings just by updating to a recent firmware, thus leading to these problems.