RMerl/asuswrt-merlin.ng #661: More sophisticated parental controls

squader, 2020-12-20:
For example, time limit access to internet for specific devices e.g. 4 hours a day only. Or limit amount of time on social websites to 2 hours a day (for specific devices).

jumpsmm7, 2020-12-20:
I am pretty sure these options are controlled by asus and not Asuswrt-Merlin. Though this is a nice feature request, I believe most of this code would be lockdown or closed source since it is maintained by Trend Micro's section of code. The only way a suggestion like this would happen is if asus collaborated with trend micro to make it happen. My only suggestion would be to bring this up as feed back to asus on their stock firmware channel.

RMerl, 2020-12-28:
There are currently no plan for me to make any change to the way parental controls are implemented, due in large part to the complexity of the code, and also because parts of it are closed source.

closed 2020-12-28 by RMerl