RMerl/asuswrt-merlin.ng #663: Solutions to multiple issues with SSL certificates

jaklan, 2020-12-24:
> Model: RT-AC86U
> Firmware Version: 386.1_beta2

Hi all, I was recently playing with SSL certificates on my router and discover a few things which imho can be helpful to solve problems with SSL certificates.

#### Generating a new self-signed cert
In `WAN - DDNS` menu, If you select `HTTPS/SSL Certificate` as `None` and `Generate a new certificate` as `Yes` then after applying - nothing changes, you still get the old certificate. The reason is `/jffs/cert.tgz` archive file, which stores 2 files within `etc` folder: `cert.pem` and `key.pem`. If the file exists, then new cert is not generated. 

Solution? Ssh into your router and run `rm /jffs/cert.tgz` before generating a new cert. Now, when applying the above, a new `/jffs/cert.tgz` file would be created and cert - successfuly replaced.

@RMerl: my opinion - removing the old file should be just done under-the-hood.

#### Importing your own cert
In `WAN - DDNS` menu, If you select `Import Your Own Certificate` then after uploading files and applying - it works (even if you get `No certificate found` status). What is interesting here - the uploaded files are stored in `/jffs/.cert/`directory, but `/jffs/cert.tgz` is not modified - it still stores the old cert files. I expected it can be a problem after reboot, but... it isn't, the cert moved to `/tmp/etc/` is still the new one. Tbh that's quite surprising to me, I need to investigate it more, but well - the most important part is "it works".

@RMerl: however it works, maybe it's a good idea to overwrite existing `/jffs/cert.tgz` anyway to have a repeatable flow?

#### Generating a certificate from Let's Encrypt (for a custom DDNS)
In `WAN - DDNS` menu, If you select `Free Certificate from Let's Encrypt` then after applying - nothing changes. It doesn't matter if you have `/jffs/cert.tgz` or any files in `/jffs/.cert/` - it just doesn't work. If you look at logs, you can see such a line: `Dec 24 02:53:29 rc_service: httpd 7936:notify_rc restart_ddns_le;restart_ftpd`, but `restart_ddns_le` seems not to do anything interesting. 

But... when you ssh into your router and run `service restart_letsencrypt`... magic happens and the process of ordering the cert starts - you can track it in `System Log` -> `General Log`. The domain used is ofc the one specified in the field `Host Name`. After all, you can `cd /jffs/.le` and you will see a new folder named as your domain (e.g. `subdomain.domain.com`).

However, we are only halfway there - the cert is not working yet. But let's do some experiment - we would get the freshly generated key and cert and pack it into `/jffs/cert.tgz`:
1) If we look at `/jffs/.le/subdomain.domain.com`, there are 2 especially interesting files for us: `subdomain.domain.com.key` and `fullchain.pem`.
2) Let's create a new `/jffs/cert.tgz` archive file:
```
cd /jffs/.le/subdomain.domain.com && mkdir etc
cp subdomain.domain.com.key etc/key.pem && cp fullchain.pem etc/cert.pem
cd tar -czf /jffs/cert.tgz etc/key.pem etc/cert.pem
```
3) Let's reboot: `reboot`
4) Voila, it works!

What next? The above doesn't answer if the cert is automatically renewed after 3 months, and even if yes - if it's moved to `/tmp/etc/` automatically. But if we you know the mechanism, you can automate it on your own or... @RMerl - again, maybe it can be automated under-the-hood, when applying Let's Encrypt cert in GUI?

PS I have tested the above for a custom domain, don't know if there are any differences if you want to generate a cert for a default *.asuscomm.com domain.

RMerl, 2020-12-28:
> @RMerl: my opinion - removing the old file should be just done under-the-hood.

erase_cert() in httpd/httpd.c wasn't deleting that backup, so it got restored just after deleting the /etc/* certs.  Should be fixed with 380955f012.

> n WAN - DDNS menu, If you select Import Your Own Certificate then after uploading files and applying - it works (even if you get No certificate found status). What is interesting here - the uploaded [...]

That archive is a backup of the active cert.  It only gets used if the router lacks any certificate to use.  It should be re-generated whenever httpd is started.

That backup is mostly redundant now that Asus also uses the /jffs/ partition to store the certificate.  I might eventually disable that code, however I need to make sure it's not referred to elsewhere like AiCloud (I know that FTP does refer to it at the moment).

> In WAN - DDNS menu, If you select Free Certificate from Let's Encrypt then after applying - nothing changes. It doesn't matter if you have /jffs/cert.tgz or any files in /jffs/.cert/ - it just doesn't work. 

Let's Encrypt is known to currently be broken due to a change in the root certificate.  Asus fixed it upstream.

> What next? The above doesn't answer if the cert is automatically renewed after 3 months,

Automatic renewal is implemented, yes.  Note that most of the LE-related code is closed source and outside of my control.  It's all handled by libletsencrypto.so.

> PS I have tested the above for a custom domain, don't know if there are any differences if you want to generate a cert for a default *.asuscomm.com domain.

Asus DDNS uses DNS-based validation, while other DDNS providers use HTTP based validation.

jaklan, 2020-12-29:
@RMerl thanks for clarification! 
> That archive is a backup of the active cert. It only gets used if the router lacks any certificate to use. It should be re-generated whenever httpd is started.
That's interesting, because even after reboot - that archive wasn't re-generated. The new own cert was still used, but backup just didn't change. Maybe it would be finally re-generated, but dunno, it was a quick test.
> Let's Encrypt is known to currently be broken due to a change in the root certificate. Asus fixed it upstream.
Oh that's great, so if I understand correctly - after merging these fixes, we would be able to create LE certs and auto-renewal them for all DDNS providers, incl. custom?

JackMerlin, 2020-12-30:
This issue has been fixed in the beta 3. I think this issue was introduced from 384.19, this is my initial report:
https://www.snbforums.com/threads/384-19-issue-unable-to-generate-a-new-certificate.67962/

and this is the fixed code:
https://github.com/RMerl/asuswrt-merlin.ng/commit/380955f0126861a32263fbbee15a2efd2098b0a5

closed 2021-01-03 by RMerl