RMerl/asuswrt-merlin.ng #665: Connection to external HD via SMB drops periodically while streaming

feranick, 2020-12-30:
Using 386.1-beta3 on RT-AC86U. External HD connected via USB 3.0 port. I have a SMB connection to it from where I stream movies. After 20-30 minutes, streaming stops, and the connection to the HD cannot be reached (from any computer/device on the network). Solution is to remove the HD from GUI and reconnect. Never had any issues using 384.xx series. 

Happy to provide more info log just let me know what it's needed.

JackMerlin, 2020-12-30:
If you have a problem with 386 beta, please feedback in the [dedicated thread ](https://www.snbforums.com/threads/asuswrt-merlin-386-1-beta-is-now-available.68205/) of the SNB forum, which will help you get faster.

feranick, 2020-12-30:
Closing this report.

closed 2020-12-30 by feranick