RMerl/asuswrt-merlin.ng #666: Adding info to ipsec.secrets doesn't work

millijuna, 2020-12-30:
I have configured my RT-AX88U (running 384.19) to bring up a set of IPSec site-to-site tunnels using StrongSwan. I achieved this by creating a simple ipsec.postconf script that appends a configuration fragment (containing the couplets for the IPSec tunnels) onto ipsec.conf. In the same script, I attempted to append the pre-shared key for the tunnels into the ipsec.secrets file using the same mechanism, but no dice. If I manually execute the command (cat /jffs/ipsec.secrets.add >> /etc/ipsec.secrets) it runs fine, but if it's in the ipsec.postconf script, no dice.

Temporary workaround is to set the global PSK to my network PSK, but I'd rather specify it for the remote VPN host directly.

Ideally, it would be nice to have .add file for ipsec.secrets configuration file, or similar.

JackMerlin, 2020-12-30:
Can you be sure that ipsec.postconf does not work? If you are not 100% sure about it, please ask your question on the SNB forums.

Please note when asking questions:
Your script may have syntax or permissions issues. Please provide a copy of your script so that others can try to help you.

millijuna, 2020-12-30:
My ipsec.postconf script does work to add the couplets to the ipsec.conf file, but does not work to add the extra secrets to the .secrets file. The sum total of the script is as follows: 

> #!/bin/sh
> cat /jffs/ipsec.conf.add >> /etc/ipsec.conf
> cat /jffs/ipsec.secrets.add >> /etc/ipsec.secrets

The first line works, the second one doesn't. If I manually execute the second line, the additional secrets get added to the bottom of the file, but of course gets lost the next time the system regenerates those files. 

To debug this, I have since added a third line that copies the ipsec.secrets file to /tmp/ (executes a simple "cp /etc/ipsec.secrets /tmp/" and the one in tmp has the additional secrets, while the one in /etc/ has since lost the additional secrets. It looks like the system is generating the secrets file after the ipsec.postconf script has run.

JackMerlin, 2020-12-30:
> My ipsec.postconf script does work

So this is not a code problem, but technical support, please report it on the SNB forums.

millijuna, 2020-12-30:
Perhaps I wasn't clear in my explanation? 

The script executes, and successfully modifies the ipsec.secrets file, but then asuswrt reverts ipsec.secrets to what would have been generated from the web interface. This appears to be an order of operations issue, with the ipsec.postconf script being called after ipsec.conf is written, but before ipsec.secrets is written by the system.

If that's intended behaviour, then please view this as a feature request to add ipsec.secrets as a file that can be modified through these scripts.

JackMerlin, 2020-12-30:
Note, I am just a user just like you, I am not a developer RMerlin.

I have been using Merlin firmware for many years. In my experience, for some scripts to runing, you need to wait, because this is the expected behavior. If you don't wait for some components to complete, your configuration will be overwritten.

I suggest you go to the SNB forum to ask your question, because it is probably not related to development.

rah66, 2021-01-24:
> My ipsec.postconf script does work to add the couplets to the ipsec.conf file, but does not work to add the extra secrets to the .secrets file.

ipsec.secrets.postconfig missing for customization

https://github.com/RMerl/asuswrt-merlin.ng/wiki/Custom-config-files

config files ***.add need place in /jffs/configs/ folder