RMerl/asuswrt-merlin.ng #667: Adaptive QoS does not limit bandwidth on AX86U using 386.1 beta 3

MyselfDeveloper, 2020-12-30:
Adaptive QoS does not limit upload and download bandwidth to specified numbers.
I get 133 Mbit/s down even if I've put the limit to 100 Mbit/s.

GeForce NOW GoS does limit.

Using a brand-new AX86U and installed 386.1 beta 3 right away.
It was working well on my old AC87U that I'm replacing, using 384.13_8.

RMerl, 2020-12-30:
Known issue, the Trend Micro engine isn't working properly in 386.1 Beta 3 for the RT-AX86U, RT-AX56U or RT-AX58U, it will require an update to a newer GPL release from Asus that includes a fixed engine.

MyselfDeveloper, 2020-12-30:
Thanks for the quick reply RMerlin.

MyselfDeveloper, 2020-12-30:
If you want me to try a daily build eventually, just ask.

JackMerlin, 2020-12-31:
I think you don't need to build it manually, because RMerlin has not got the repair code from Asus yet. Even if the code is fixed, it takes a lot of time to merge it, so just wait and subscribe to [this thread](https://github.com/RMerl/asuswrt-merlin.ng/issues/667) in the SNB forums.

MyselfDeveloper, 2021-01-02:
FYI: 
It seems that Adaptive QoS works on LAN ports in Asus Version 3.0.0.4.386.41535 (AX86U).
However, it does not work on 5GHz radio. (no idea if it works on 2.4GHz).

It's a shame (on ASUS) that Adaptive QoS does not work. 
Once you have used good QoS, you cannot live without.
So, I'm back with my AC87U as the router and my AX86U as an access point.

Is there anyway I can put pressure on ASUS on that issue?
@JackMerlin, your link points to GitHub and not to SNB forums.

RMerl, 2021-01-02:
There is no need to apply pressure on Asus, we are still waiting for them to prepare updated GPL drops with an updated engine.  Just relax, this is a BETA.

MyselfDeveloper, 2021-01-02:
OK. Thanks RMerlin.

JackMerlin, 2021-01-03:
> your link points to GitHub and not to SNB forums.

Thanks for the reminder, I found that the clipboard tool I am using always makes mistakes, I may have to give up using it, the link has been corrected, thank you.

MyselfDeveloper, 2021-01-09:
(also posted in the forum: https://www.snbforums.com/threads/asuswrt-merlin-386-1-beta-stage-2-is-now-available.69116/page-5#post-649725)

Hi Guys,

I tested Adaptive QoS on AX86U using the new Beta 4.

The downstream QoS is limiting bandwidth fine on all interfaces : Ethernet, 2.4GHz and 5Ghz.
**The upstream QoS is not limiting bandwidth on 2.4GHz and 5Ghz.** It works only on the Ethernet ports.

----
The details:
- Internet connection 135/23 as tested without QoS.
- QoS settings : Adaptive QoS, Manual speed, Max download: 100, Max upload: 20.
- SpeedTest on Ethernet : 98.58 down, 19.75 up
- SpeedTest on 5Ghz : 98.47 down, **22.82 up**
- SpeedTest on 2.4Ghz : 98.44 down, **22.82 up**

RMerl, 2021-01-28:
Closing this, as this is outside of my control.

closed 2021-01-28 by RMerl