RMerl/asuswrt-merlin.ng #668: Apple M1 issues

lorenc-tomasz, 2020-12-31:
Hi,
I was playing with new Apple M1 hardware and I have noticed that the WiFi shows as connected, but it doesn't seem to send any traffic over the network. I can't ping the router -- it just times out. The Tx Rate in the wifi menu fluctuates a little bit, which makes me think it's sort of connected, but no traffic goes through. Turning wifi off and on again fixes it immediately. No other devices I have have ever had a similar issue. Is there any M1 <> ASUS incompatibility? 

More: https://forums.macrumors.com/threads/wifi-issues-m1-macbook-pro.2270884/

JackMerlin, 2020-12-31:
Please ask Apple, because Asus routers use industry standards, if there is no problem with other devices, it is not a router problem.

So I suggest you go to Apple support forums or just call their genius. that's will be very helpful.

lorenc-tomasz, 2020-12-31:
They only suggest to restart the device ;) I thought that there is maybe some know issue with their new Apple Silicon. As the old Intel machines work properly :) 

Thank you for answer and Happy New Year :))

JackMerlin, 2020-12-31:
Check [this guide](https://github.com/RMerl/asuswrt-merlin.ng/issues/668) posted by RMerlin, and if you still encounter problems, I think we will be powerless, sorry.

You too, hope you can close this non-asus issue.

trajano, 2021-01-01:
I don't think it's limited to M1, I have an 2020 Intel MacBook Pro and I am also unable to connect.  I am presuming it is because of AiMesh do you have a similar set up at your place?  If so the way I resolved it was to create a guest network that has access to the intranet and that works well with the Mac.

lorenc-tomasz, 2021-01-01:
@JackMerlin The url is to this issue not to guide. 

@trajano I don't use AiMesh. Only one AX88U setup. Some people wrote that switching to tp-link resolved the issue for them but it's strange as the mentioned tp-link uses the same chipset as the AX88U. The only guess for me is that they use different wifi modules and there is some driver issue. However, on the other hand, with the AC87U the issue doesn't occur.

trajano, 2021-01-01:
Try creating a guest network to see if it will connect to that one then, maybe it's not aimesh per se but the use of a secondary guest network that fixed my issue.

JackMerlin, 2021-01-01:
> @JackMerlin The url is to this issue not to guide.
>
> @trajano I don't use AiMesh. Only one AX88U setup. Some people wrote that switching to tp-link resolved the issue for them but it's strange as the [...]

Thanks for your notification, I have fixed it

lorenc-tomasz, 2021-01-01:
@trajano I have tried the guest network and the result is similar when I have disabled the WIFI6. The wifi connection stayed longer but finally after few minutes all network traffic stopped. I have tested it with Hotspot created on iPhone, and there is no issue. I also borrowed the tp-link archer ax6000. And with TP-link I had no issues at all. So it's something Apple<>Asus related. So I have to solutions - change router or try to use usb dongle. Probably I will try the second option as I like my AX88U a lot :)

trajano, 2021-01-01:
Another thing you can try is to switch to official firmware on your Asus router.  Though that may be a bit too drastic.  I do see there's a new beta of AsusWRT https://www.snbforums.com/threads/asuswrt-386-rc2-public-beta-with-full-functions-aimesh-2-0.65235/ maybe there'll be an update of the custom firmware for this one.

lorenc-tomasz, 2021-01-01:
Beta doesn't help. At the moment I can say that,

M1 doesn't work at the moment with:
- ASUS AX88U
- ASUS TUF Gaming AX3000
- ASUS RT-AC86U

M1 works OK with:
- ASUS AC87U
- ASUS RN16N
- TP-Link Archer AX6000
- iPhone 12 AX hotspot

At the moment I don't have access to more routers.

lorenc-tomasz, 2021-01-01:
OK. Super strange - I have connected through VPN on the routers where this issue occurs. And there is no issue with VPN connection. So it's suggesting some routing problem on M1.

trajano, 2021-01-02:
I just upgraded to Beta3 on the AC68U and my connection issues are gone from the Macbook,.  I am having some weirdness with the Pi4 though still checking it out, I think I may just reset my pi.

lorenc-tomasz, 2021-01-02:
I did clean beta3 install on AX88U. The same result - traffic is dropped. When using VPN there is no issue at all. It's super strange. On AX88U I need vpn to not lost connection. On TP-link it's working without issues. Hope new Wifi dongle will resolve the issue.

hurda, 2021-01-02:
Do you have an Apple Watch?
Have you done some tcpdumps with Wireshark or similar?

lorenc-tomasz, 2021-01-02:
Yes, I have Apple Watch but it works ok. Only M1 has problems. 

I ran Wireshark and was able to reproduce the disconnection issue by installing Xcode from the App Store. Downloads for awhile and then stops. It's weird, though -- traffic still seems to come in on the interface. Mostly mDNS broadcast requests from other devices on the network. I can see my outgoing ping requests to my router but they don't get a response. I also see a fair number of TCP retransmissions from the M1 to destination.

screenshots: 
- https://ibb.co/100xWCS
- https://ibb.co/dBrnWBm

RMerl, 2021-01-03:
Anything related to wifi is outside of my control, so closing this.  Forums are a better place for such discussions.

closed 2021-01-03 by RMerl