RMerl/asuswrt-merlin.ng #672: Unable to load admin page AC88U 386.1 beta3

achampion, 2021-01-03:
Just loaded beta-3 to an AC88U, ow the admin page is not accessible.
It prompts for username and password but then timeouts for both HTTPS and HTTP (local network).

I can still access it via ssh, what is the command line to install the old firmware?
    -A.

RMerl, 2021-01-03:
Please read the Changelog, your answer is there...

closed 2021-01-03 by RMerl

JackMerlin, 2021-01-03:
Interesting, did you see the issue template?

JackMerlin, 2021-01-04:
https://github.com/RMerl/asuswrt-merlin.ng/blob/master/Changelog-NG.txt
````
  - NOTE: Some users upgrading might have to go through some
          database maintenance on first boot, which means the
          router might be slower or have a non-responsive webui
          for a while.
          This can take anywhere from 5 minutes up to an hour,
          depending on your model, just give it time to complete
          the process.
````
https://www.snbforums.com/threads/asuswrt-merlin-386-1-beta-is-now-available.68205/
````
Upgrade notes:
- If coming from a previous alpha build, you MUST do a factory default reset after flashing this beta firmware.
- If coming from stock Asus firmware, a factory default reset is recommended, but not mandatory.
- If going back to stock Asus firmware, a factory default reset is STRONGLY recommended.
- If updating your GT-AC2900 from an older 384_xxxx firmware, reformatting your JFFS partition is STRONGLY recommended.
- Direct upgrade from 384.18 or 384.19 should be fine, but be prepared to do a factory default reset if something does not work as expected.
````

If you have other questions about the 386.1 beta, please discuss them in [this thread](https://www.snbforums.com/threads/asuswrt-merlin-386-1-beta-is-now-available.68205/).
Thank you.