RMerl/asuswrt-merlin.ng #673: RT-AC3100 et firmware 384.19 : No broadcast between 2.4 Ghz and 5 Ghz

llooping, 2021-01-04:
Hi,

I already ask on the community forum and nobody replies.

Here my example of the issue of my Wifi : 
I have a device connected on the 2.4 Ghz, for example a TV but works with any device like that
I have another device connected on the 5 Ghz, my smartphone or any device
I launch my TV app on my smartphone (on 5 Ghz) to connect to my TV (on 2.4 Ghz)
The app say there is no device on the network
I force my smartphone to connect to 2.4 Ghz like my TV and it works, the app find and connect to the TV
If I change, I force my TV and mt smartphone on the 5 Ghz, it works too.
I have this issue with all devices wich need to connect throught 1 device on 2.4 Ghz and 1 device 5 Ghz, I have this issue.

Router Asus RT-AC3100 with 384.19 with the same WIFI SSID for both Channel (I let the device to connect to the right channel 2.4 or 5)

Thanks

JackMerlin, 2021-01-04:
Please use the template to ask questions, and carefully read the comments in the template.

JackMerlin, 2021-01-04:
> I already ask on the community forum and nobody replies.

You need to understand that everyone in this open source community has a full-time job. This is just created by them in their spare time. They have no way to provide 24/7 support. If you don't want to wait, there will never be an answer.

llooping, 2021-01-04:
Done, Thx

JackMerlin, 2021-01-04:
> 
> 
> Done, Thx

Don’t delete the template, I don’t see Expected behavior and Actual results

JackMerlin, 2021-01-04:
> I already ask on the community forum and nobody replies.、

This may be a technical support issue, I will hand it over to the community to try to solve it, but first please provide your thread link. Also, please note that I am not RMrlin, I am just a user like you, trying to help this community.

llooping, 2021-01-07:
OK thanks

JackMerlin, 2021-01-09:
> 
> 
> OK thanks

If you have already asked the question in the forum, please close this question in here, thank you.

dkobelak, 2021-01-10:
this happens to me sometimes as well.. usually when cellphone is using 5ghz and a device such as sonos is on 2.4ghz.. i have to switch the phone to 2.4 ghz.. it's rare this happens, but it is annoying when it does.. it would be nice to have the link to the forum thread to see what is being said about this..

llooping, 2021-01-10:
I agree we are not alone with this issue.......but I feel nobody care about this issue........:(

When I have time, I will test with the official release to check if it's a issue from ASUS or Merlin.

RMerl, 2021-01-10:
I don't care about it simply because I have no control over anything related to wifi.  Reproduce the issue with stock firmware, then report it to Asus.

closed 2021-01-10 by RMerl