RMerl/asuswrt-merlin.ng #674: In repeater mode, NTP function does not seem to work

testn, 2021-01-05:
In repeater mode, NTP function does not seem to work. The time 

**Model**
RT-AC68U

**Firmware version**
384.19

**Client information**
Chrome 87

**To Reproduce**
Check the time on the router. (https://<ip>:8443/Main_LogStatus_Content.asp) 

**Expected behavior**
The time should be the current time

**Actual results**

It will show the time in 2018

System Time | Tue May 8, 2018
Uptime | 2 days 19 hour(s) 1 minute(s) 3 seconds

**Will this error be reproduced on the official firmware?**
I didn't test

**Screenshots**

**Additional context**

JackMerlin, 2021-01-05:
Have you ever tried to manually click the NTP LINK on the admin page?

testn, 2021-01-19:
Already tried and enabled the network monitoring but it still does not work.

RMerl, 2021-03-20:
Tested, working in 386.2 Beta 1.

closed 2021-03-20 by RMerl

testn, 2021-03-26:
Thanks @RMerl  I will check it out.