RMerl/asuswrt-merlin.ng #675: GoLang ARMv7 binaries throw "Illegal Instruction" error on ARMv7 CPU. VFP disabled

alexsarmiento, 2021-01-05:
```
# nvram get model
RT-AC68U

# cat /proc/cpuinfo
Processor       : ARMv7 Processor rev 0 (v7l)
processor       : 0
BogoMIPS        : 1595.80

processor       : 1
BogoMIPS        : 1599.07

Features        : swp half thumb fastmult edsp
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x3
CPU part        : 0xc09
CPU revision    : 0

Hardware        : Northstar Prototype
Revision        : 0000
Serial          : 0000000000000000
```

_Firmware Version:386.1_beta3_

When trying to install AdGuardHome armv7 on this router I encountered the aforementioned error. It seems that GoLang ARMv7 (and ARMv6) builds require VFP support from the kernel which seem disabled:

`# CONFIG_VFP is not set`

Granted, GoLang ARMv5 builds do work, but such a shame .  I don't know how wise if possible to enable this kernel feature but it might somewhat prolong the longevity and usefulness of the router.

JackMerlin, 2021-01-05:
Please use the template to report error, and carefully read the comments in the template.

RMerl, 2021-01-05:
Nothing I can do about it if the CPU hardware does not support it.

closed 2021-01-05 by RMerl