RMerl/asuswrt-merlin.ng #676: Random clients cannot connect to network.

ServerGuyHK, 2021-01-14:
Not sure if I should give Beta feedback here, but here goes.

RT-AX88U running in AP mode with a RT-AC68U as an AiMesh Node.
DHCP and DNS are running over two Windows 2019 Servers with load balancing.
The AX88U is plugged directly in to a Netgate Firewall running pfSense.
The AC68U is hard wired to the AX88U for its backhaul for the AiMesh network.

Running the latest stock firmware, all works and clients can connect to both 2.4 and 5GHz networks without issue and get an IP from the DHCP servers.

I installed:
RT-AX88U_386.1_beta3
RT-AC68U_386.1_beta4

Once installed I could see that random devices (Nest smoke alarm, Sensibo AC Control, HP Printer etc) could not attached to the network. Even trying to reset the network config on the devices did not work, they just would not connect.

Likewise my guest network stopped working and devices could not connect to this.

Reinstalling:
RT-AX88U_3.0.0.4_384_9579-g5362374
RT-AC68U_3.0.0.4_386_40558-gc48f410

Back on the routers without changing a line of config and everything starts working again.

All I can think of is these devices cannot IPs from DHCP, but I am at a loss as how to prove this.

JackMerlin, 2021-01-14:
Thank you for your feedback, but I think there may be no one here to help you, because this question may not seem to be a bug.

I suggest you ask your questions to [our support community](https://www.snbforums.com/forums/asuswrt-merlin.42/). And hope you follow our bug template the next time you submit any bugs.

Note that I am not a developer RMrlin, I am just a user.

Best regards
Jack

JackMerlin, 2021-01-14:
If you understand the answer above, you can close this question to help this project.

ServerGuyHK, 2021-01-14:
Hi, 
I have posted on the community forum as suggested, but I would have though this is a bug.
Something as fundamental as DHCP not working on the latest beta, but working on the stock firmware.

I will close this then, if thats not considered a bug.

Thanks

closed 2021-01-14 by ServerGuyHK

JackMerlin, 2021-01-15:
> Hi,
> I have posted on the community forum as suggested, but I would have though this is a bug.
> Something as fundamental as DHCP not working on the latest beta, but working on the stock firmware.
> 
> I will close this then, if thats not considered a bug.
> 
> Thanks

A real bug must be able to be reproduced, and the information you provide is not enough to help others reproduce the problem. If it is just an urban legend, then we will not regard it as a bug.

And, please provide your thread link to help other people with the same problem search for possible solutions.

Thanks

ServerGuyHK, 2021-01-18:
Hi,

I dont think this is an urban legend and yes I can reproduce it by reinstalling the betas back on to my two routers and just waiting for devices to need new DHCP addresses. I can fix it by returning to the stock firmware. I dont need to change any config or do anything new, just leave the settings as they are and swap around the config.

This was my thread:
www.snbforums.com/threads/random-clients-cannot-connect-to-network-dhcp-cannot-be-contacted.69298/#post-652856

But I have since found this very large thread that seems to have the same issues when you read through.
www.snbforums.com/threads/in-some-dire-need-of-assistance-ax86u.67944/page-4#post-652324

Thanks again.