RMerl/asuswrt-merlin.ng #677: Slow bitrate on router, no in LAN

jellyfinuser, 2021-01-18:
**Describe the bug**

My router is asus RT-AC68U. I have an Azure Server as frontend, and one Rpi4 in my home with wireguard server to permit Azure server access to my lan (Nextcloud). Testing multiple configs to try to get more bandwith I saw interesting situation:

1.- From azure server to iperf server on asus (lan side) I get arround 50 Mb/s

[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  4.85 MBytes  40.7 Mbits/sec    0    294 KBytes
[  5]   1.00-2.00   sec  7.79 MBytes  65.3 Mbits/sec    0    381 KBytes
[  5]   2.00-3.00   sec  8.03 MBytes  67.4 Mbits/sec    0    381 KBytes
[  5]   3.00-4.00   sec  7.30 MBytes  61.2 Mbits/sec    0    381 KBytes
[  5]   4.00-5.00   sec  8.03 MBytes  67.4 Mbits/sec    0    381 KBytes
[  5]   5.00-6.00   sec  8.03 MBytes  67.4 Mbits/sec    0    381 KBytes
[  5]   6.00-7.00   sec  7.17 MBytes  60.2 Mbits/sec    0    381 KBytes
[  5]   7.00-8.00   sec  8.22 MBytes  68.9 Mbits/sec    0    381 KBytes
[  5]   8.00-9.00   sec  7.30 MBytes  61.2 Mbits/sec    0    381 KBytes
[  5]   9.00-10.00  sec  8.03 MBytes  67.4 Mbits/sec    0    381 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  74.8 MBytes  62.7 Mbits/sec    0             sender
[  5]   0.00-10.00  sec  73.2 MBytes  61.4 Mbits/sec                  receiver

2.- From Azure server to RPI Iperf Server I get 300 Mbits/seg (all my isp bitrate)

[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  26.5 MBytes   222 Mbits/sec    0   2.53 MBytes
[  5]   1.00-2.00   sec  33.8 MBytes   283 Mbits/sec    0   2.60 MBytes
[  5]   2.00-3.00   sec  35.0 MBytes   294 Mbits/sec    0   2.60 MBytes
[  5]   3.00-4.00   sec  35.0 MBytes   294 Mbits/sec    0   2.60 MBytes
[  5]   4.00-5.00   sec  35.0 MBytes   294 Mbits/sec    0   2.60 MBytes
[  5]   5.00-6.00   sec  35.0 MBytes   294 Mbits/sec    0   2.60 MBytes
[  5]   6.00-7.00   sec  35.0 MBytes   294 Mbits/sec    0   2.60 MBytes
[  5]   7.00-8.00   sec  35.0 MBytes   294 Mbits/sec    0   2.60 MBytes
[  5]   8.00-9.00   sec  33.8 MBytes   283 Mbits/sec    0   2.60 MBytes
[  5]   9.00-10.00  sec  36.2 MBytes   304 Mbits/sec   12   2.60 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   340 MBytes   285 Mbits/sec   12             sender
[  5]   0.00-10.07  sec   339 MBytes   283 Mbits/sec                  receiver

**Model**
My router is asus RT-AC68U 

**Firmware version**
firmware version 384.19

**Client information**

**To Reproduce**

**Expected behavior**
More bitrate at router or similiar bitrate when I test on RPI iperf server or router iperf server

**Actual results**

**Will this error be reproduced on the official firmware?**

**Screenshots**

**Additional context**

JackMerlin, 2021-01-18:
Hello, @jellyfinuser 

You must tell us how to reproduce this error. If you do not tell us, we cannot help you.

Please check the question template carefully, it is good for you

Thank you

jellyfinuser, 2021-01-18:
Edited, I hope now will be clear.

Thanks.

JackMerlin, 2021-01-18:
> Edited, I hope now will be clear.
Hello,
No one can help you with non-software error cases. If you are not sure that this is a software error, please close this question. If you are sure, please provide a detailed proof, because you are always ambiguous.

It seems to me that your router hardware limits the forwarding performance of switch. so, my suggestion is to try to restore the factory reset and see if the same result is obtained.
If it is, try to flash to the official firmware to see if the result is the same.
If so: You are questioning the performance of your home router, so try to choose enterprise-grade equipment.

In addition, I am not a developer, I am just a user.

Thanks

RMerl, 2021-01-28:
iperf3 requires a fair amount of CPU, your bottleneck is iperf and the outerCPU here.  Not a bug.

closed 2021-01-28 by RMerl