RMerl/asuswrt-merlin.ng #678: Upgrade dnsmasq to 2.83

thejohnha, 2021-01-21:
Thank you so much for all your hard work, Merlin :-)

I saw this come across and I'm sure it's on your radar, but I thought I'd check:
https://us-cert.cisa.gov/ncas/current-activity/2021/01/21/certcc-and-cisa-report-multiple-vulnerabilities-dnsmasq

I see that the latest version of dnsmasq is 2.83 which would mitigate these vulnerabilities.

Thank you again!
-John

RMerl, 2021-01-21:
Already aware of it, yes.  Unfortunately 2.83 is broken and currently unusable, we have to wait for the dnsmasq author to fix it.  See this discussion:

https://forum.openwrt.org/t/security-advisory-2021-01-19-1-dnsmasq-multiple-vulnerabilities/85903/22

themiron, 2021-01-26:
updated to 2.84 with 158c209f23f308538b6e7a4b907bb459209da4df

closed 2021-01-26 by themiron