RMerl/asuswrt-merlin.ng #680: Improve WebUI Switch Screen Reader Accessibility

Flameborn, 2021-01-22:
The WebUI is mostly accessible to screen reading software, such as [NVDA](https://nvda-project.org) under Windows, the built-in VoiceOver on MacOS and iOS, as well as Orca under Linux. The only issue seems to be the on/off switches, which are based on a *div* element, as well as an image. These can be found, for instance, on the "QoS - WAN/LAN Bandwidth Monitor" page.

Screen readers are unable to determine the state of these switches, e.g. VoiceOver speaks them as "iphone_switch_container_on.png image", regardless of their state.

Each *div* elements should be marked with an ARIA switch role, as well as have an associated *tabIndex* attribute (e.g. 0), so that they can be navigated to via the keyboard.

When they are focused, a *keyPress* event should check if the spacebar is pressed, and toggle the switch state, similarly to how click events are handled.

More info on the ARIA switch role can be found [here](https://developer.mozilla.org/en-US/docs/Web/Accessibility/ARIA/Roles/Switch_role).

Hopefully this small improvement can be added upstream as well.

RMerl, 2021-01-28:
Unfortunately such wide ranging webui changes aren't really realistic as it would make very difficult for me to merge any future webui changes from Asus.  Those types of changes are better handled upstream by Asus themselves.

Flameborn, 2021-01-28:
Makes sense. I'm not sure how willing Asus is to improve this though, nor about the time it would take them to implement it.

Is there a way to forward this as a request to them, or should I go through their support as usual?

RMerl, 2021-01-28:
If you run their original firmware, there is a feedback page on the router that you can use to contact them.  Otherwise, email them at networking_support <at> asus <dot> com.

closed 2021-01-28 by RMerl