RMerl/asuswrt-merlin.ng #682: 386.1 regression: Auto Logout is not working

bughit, 2021-02-02:
**Describe the bug**
browser remains logged in indefinitely

**Model**
RT-AC68U

**Firmware version**
386.1_0 upgraded from 384.19_0

**Client information**
Firefox 85

**To Reproduce**
- set auto logout to 10 minutes
- close the router tab
- after 10+ minutes load a router page

**Expected behavior**
auto logout

**Actual results**
no auto logout

paldier, 2021-02-02:
You have to clear your cookies, or login on another device, or change your username and password, everyone can use your cookies to access your router, it's not a bug, because you didn't click logout

snaunton, 2021-02-04:
@paldier - isn't the point of auto logout that you get logged out automatically after a certain period of time even if you did not click logout?

bughit, 2021-02-14:
I may be misremembering how auto-logout works. I tried it in 384.19_0 and it only logs out if a router page is kept open. One would expect it to expire the session regardless. So either it never worked right or it regressed earlier.