RMerl/asuswrt-merlin.ng #683: 386.1 Time Scheduling in Parental Control does not work

hitsukeyaku, 2021-02-02:
**Describe the bug**

When using the 'Time' type for Parental control, it allows access even if the proper offline hours are set.

**Model**

RT-AC68U

**Firmware version**

386.1

**Client information**

Chrome Version 87.0.4280.141 (Official Build) (x86_64)

**To Reproduce**

1. Go to AI Protection -> Parental Controls -> Time Scheduling
2. Add a device (Select a client name in the pull down menu)
3. For the added device, select "Time" and edit  the entry for the device by clicking on the edit icon (pencil icon) for Time Management.
4. Input appropriate Time Schedule for the offline time.
5. Click on Apply
6. In the Time Scheduling page, click on Apply again

**Expected behavior**

Device should be prevented from accessing the internet during the given offline schedule.

**Actual results**

Device can still access the internet.

**Will this error be reproduced on the official firmware?**

Yes

**Screenshots**

**Additional context**

RMerl, 2021-02-02:
This will have to be fixed upstream by Asus.

snaunton, 2021-02-05:
It looks like the --kerneltz parameter to iptables is borked. If use iptables rules directly, e.g.

`iptables -I FORWARD 1 -i br0 -m time --timestart 11:00:00  --timestop 19:59:59 -m mac --mac-source <MACADDRESS> -j DROP`

Then it seems to work ok (in UTC timeszone).

If I then use:

`iptables -I FORWARD 1 -i br0 -m time --timestart 11:00:00  --timestop 19:59:59 --kerneltz -m mac --mac-source <MACADDRESS> -j DROP`

The results seem to be unpredictable.

Note that timescheduling basically uses --timestart ---timestop and --kernaltz.

I also noticed that the local time zone (/etc/TZ) is set wrong (at least for Sydney, Australia). 

cat /etc/TZ:

`UTC-10DST`

The date command returns non-DST time, e.g.

`Fri Feb  5 21:03:36 UTC 2021`

 If I echo UTC-10DST,M10.1.0/2,M4.1.0/3 > /etc/TZ (correct setting for Sydney, Australia) then date returns correctly:

`Fri Feb  5 22:03:36 DST 2021`

I never noticed this before 386.1, but I don't think this date thing was a problem. 

I suspect both are related.

RMerl, 2021-02-12:
Should be fixed with 858b57b2b7 .

closed 2021-02-12 by RMerl