RMerl/asuswrt-merlin.ng #688: Router itself can't use IPv6 though LAN clients can

dee-kryvenko, 2021-02-05:
**Describe the bug**

The router itself doesn't seem to use IPv6 properly, though all the clients on LAN can. I'm not 100% sure it's not a configuration issue - but it looks like one. All external actions like checking for firmware updates and registering LetsEncrypt cert taking ages and normally I wouldn't probably notice, but then I tried installing Entware and noticed a lot of `connection timeout` errors in the logs. It seems like underlying curl/wget tries to use IPv6 address first and gets a timeout (which takes considerable time) before falling back to IPv4.
I tried `ping -6 2001:4860:4860::8888` and `traceroute6 2001:4860:4860::8888` and both fails, though LAN clients on network works just fine. Upstream AT&T router is configured as IP passthrough.

**Model**

RT-AX86U

**Firmware version**

386.1

**Client information**

**To Reproduce**

[image]

**Expected behavior**

Router should be able to use IPv6 itself

**Actual results**

Router fails to use IPv6

**Will this error be reproduced on the official firmware?**

**Screenshots**

**Additional context**

maghuro, 2021-03-07:
Using native ipv6 also, and everything looks good.

I can ping and traceroute the ipv6s you provided.

dee-kryvenko, 2021-03-07:
Any tips for further diagnostic?

hktt, 2021-07-10:
I have the same issue with 386.2_6. Please provide a resolution. Running entware / amtm updates take much longer because IPv6 can not connect and then it takes time to fall over to IPv4.

RMerl, 2021-07-10:
Things related to IPv6 will have to be done upstream by Asus, as I have no way of testing it, and I don't expect my ISP to support IPv6 for many years.