RMerl/asuswrt-merlin.ng #689: 386.1 Speed Test tops out at 500 Mbps on a 900 Mbps PPPoE Internet connection

dbmajor, 2021-02-07:
**Describe the bug**
RT-AX88U Speedtest under QoS tab only reports 500 Mbps download speed, when the connection speed is 900 Mbps and tests at that on a laptop connected via Ethernet.

**Model**
RT-AX88U

**Firmware version**
386.1_0

**Client information**
The router itself

**To Reproduce**
Connect RT-AX88U to a PPPoE internet connection and attempt a speed test under the QoS section of the router.

**Expected behavior**
A speed test with a download of 900-960 Mbps download and 500 Mbps upload

**Actual results**
A speed test with a download of 500 Mbps download and 500 Mbps upload

**Will this error be reproduced on the official firmware?**
Have only tested on Merlin Firmware

**Screenshots**
See attached

**Additional context**
I'm running Mac OS X Catalina, but can't see how that would affect results, as the speedtest website reports ~930 Mbps

crkinard, 2021-02-07:
Hardware limitations of the router. Was listed in the release notes.

RMerl, 2021-02-07:
That would be a hardware limitation.

closed 2021-02-07 by RMerl