RMerl/asuswrt-merlin.ng #698: SSL for webconfig doesn't use imported certificates when upgraded to 386.1

tedd-wq, 2021-02-12:
**Describe the bug**
I have my own SSL certs. They worked in 384.19, but webconfig reverted to self-signed certificates and every attempt to delete these certs from /etc and re-apply my own fails, with newly generated self-signed certs again. My certs get uploaded to /jffs/.cert, but from there, they get ignored.

**Model**

RT-AC88U

**Firmware version**
386.1

**Client information**
Vivaldi, latest version (can't check currently). Also Firefox 85.0.2.

**To Reproduce**

1. Disable https webconfig
2. `cd /etc`
3. `rm cert.crt cert.pem key.pem server.pem` (or not. Doesn't matter.)
4. Use webconfig to upload certs to router
5. Enable https or both auth method.
6. Webconfig restarts, using self-signed certs.
7. Self-signed cert details on DDNS page.
8. Turn off https auth method (go back to http-only)
9. My uploaded cert details shown on DDNS page.

**Expected behavior**

As above

**Actual results**

As above

**Will this error be reproduced on the official firmware?**

Not tested

**Screenshots**

https://imgur.com/a/9Jh81Ih

**Additional context**

My cert and chain are ECDSA, and the private key is EC, but that shouldn't cause the problem, as they were working in 384.19 and previous.

I fiddled with nvram https_crt_save set to 1 and 0 during this. I don't think I had to do anything like this in 384.19 (or whenever I initially started using these certs).

crkinard, 2021-02-12:
I would like to second this.

Even uploading it through the web interface does not use the SSL cert/key you upload.

RMerl, 2021-02-12:
Working for me.  I went to the DDNS page, uploaded a new key and certificate I just generated, clicked Apply.  Closed and reopened browser, and next login was using the new certificate.

Try with a regular RSA key instead of EC.

tedd-wq, 2021-02-13:
Did some experimenting with different key algorithms. Installing an RSA 4096 cert and key works, using my same EC intermediate and root CA certs.

I would class this as a regression as I've had a prime256v1 key and associated certificate installed since at least 384.19. Is this something you can fix @RMerl, or are we stuck with this because it's from ASUS?

registered-to-post-this, 2021-02-20:
This issue also occurs on the latest official firmware for the AC68U, 9.0.0.4.386_41994. Uploading an EC certificate and key from the DDNS page will silently fail and the last used RSA certificate and key will be retained.

In fact, uploading certificates in this version of the firmware seems to be broken entirely. Setting "Authentication Method" to either "HTTPS" or "BOTH" will result in a self-signed router.asus.com certificate being installed, and from then on any certificate/key of any type uploaded from the DDNS page is simply ignored.

All certificates used were signed directly by a self-created CA.

edgan, 2021-04-04:
I ran into this yesterday on my AC5300 while trying to use a LetsEncrypt certificate. I generated a new certificate, and it generated a new EC private key. I spent hours trying to figure out why the router kept generating a new certificate and key when I put my certificate and key in place. I finally realized it must not like EC private keys. I then generated a RSA private key, and it just worked. I think the issue is with the mssl_cert_key_match function that is testing if the certificate and the key match each other.

Lloyd1978, 2021-04-19:
> 
> 
> Working for me. I went to the DDNS page, uploaded a new key and certificate I just generated, clicked Apply. Closed and reopened browser, and next login was using the new certificate.
> 
> Try with a regular RSA key instead of EC.

I've had the same problem with all the versions of 386 on my RT-AC86U, going back to 384.19 fixes the issue. As does using a RSA certificate on 386.2_2.
Regards

Lloyd

RMerl, 2021-04-19:
EC certificates were never officially supported.  The fact that it used to work was just a coincidence, and it stopped working once Asus implemented certificate validation with 386_xxxxx.

Adding EC certificate support will require extra work, and unless someone can provide with a patch, this has a very low priority at this time.