RMerl/asuswrt-merlin.ng #699: (GMT+10) Canberra, Melbourne, Sydney timezone (and perhaps others) actual bug

snaunton, 2021-02-13:
**Describe the bug**

nvram variable time_zone_x, therefore /etc/TZ is not set property for Sydney timezone. This is caused by a bug in the C code.

**Model**

RT-AC68U

**Firmware version**

386.1_2

**Client information**

**To Reproduce**

Set to (GMT+10) Canberra, Melbourne, Sydney timezone in Administration system.

**Expected behavior**

date command should return correct time.

`# nvram show | grep ^time_zone:`
time_zone_dst=1
time_zone_x=UTC-10DST,M10.1.0/2,M4.1.0/3
time_zone_dstoff=M10.1.0/2,M4.1.0/3
time_zone=UTC-10DST_1
# cat /etc/TZ:
UTC-10DST,M10.1.0/2,M4.1.0/3`

**Actual results**

date command is 1 hour behind during DST period.

`# nvram show | grep ^time_zone:
time_zone_dst=1
time_zone_x=UTC-10DST
time_zone_dstoff=M10.1.0/2,M4.1.0/3
time_zone=UTC-10DST_1
# cat /etc/TZ:
UTC-10DST`

**Will this error be reproduced on the official firmware?**

Have not tested

**Screenshots**

**Additional context**

This seems to be caused by a bug in void time_zone_x_mapping(void) in release/src/router/rc/common.c:

release/src/router/rc/common.c:

void time_zone_x_mapping(void)
...
	len = snprintf(tmpstr, sizeof(tmpstr), "%s", nvram_safe_get("time_zone"));
	/* replace . with : */
	while ((ptr=strchr(tmpstr, '.'))!=NULL) *ptr = ':';
	/* remove *_? */
	while ((ptr=strchr(tmpstr, '_'))!=NULL) *ptr = 0x0;

	/* check time_zone_dst for daylight saving */
	if (nvram_get_int("time_zone_dst"))
		len += sprintf(tmpstr + len, ",%s", nvram_safe_get("time_zone_dstoff"));
...
	if ((fp = fopen("/etc/TZ", "w")) != NULL) {
		fprintf(fp, "%s\n", tmpstr);
		fclose(fp);
	}

The first line of the above copies "UTC-10DST_1" to tmpstr:

len = snprintf(tmpstr, sizeof(tmpstr), "%s", nvram_safe_get("time_zone"));

... and len is set to the length of "UTC-10DST_1", i.e. 11
                                       
Then the line ...

while ((ptr=strchr(tmpstr, '_'))!=NULL) *ptr = 0x0;

... sets tmpstr to "UTC-10DST\01", so the actual null terminated string is now "UTC-10DST", but there is a "1" after the null.

For sydney time_zone_dst is 1 so the following is true:

if (nvram_get_int("time_zone_dst"))

So the following line is executed:

len += sprintf(tmpstr + len, ",%s", nvram_safe_get("time_zone_dstoff"));

However, len is still the length of UTC-10DST_1, not UTC_10DST so tmpstr will be set to:

"UTC-10DST\0,1M10.1.0/2,M4.1.0/3"

Because of the null terminator the string is actually just "UTC-10DST".

len should be the length of UTC-10DST, i.e. 9, before appending time_zone_dstoff so tmpstr is set to:

"UTC-10DST,M10.1.0/2,M4.1.0/3"

Something like the following should fix the problem:

	while ((ptr=strchr(tmpstr, '_'))!=NULL) *ptr = 0x0;

        /* recalculate length of tmpstr after potentially modifying it */
        len = strnlen(tmpstr, sizeof(tmpstr));

	/* check time_zone_dst for daylight saving */
	if (nvram_get_int("time_zone_dst"))
		len += sprintf(tmpstr + len, ",%s", nvram_safe_get("time_zone_dstoff"));

Sorry, I don't have time to compile the whole thing and install and test all timezone combinations, but I hope this bug report points you in the right direction.

Trainax, 2021-02-17:
Is this issue the same as #695?

snaunton, 2021-02-17:
Yes. My post that this was an asus thing was wrong. I found the bug and fixed it.

closed 2021-03-08 by RMerl