RMerl/asuswrt-merlin.ng #700: RT-AX86U_386.1_2 "Invalid Firmware Upload"

WhiteZeroX, 2021-02-15:
**Describe the bug**
Just got my AX86U in and trying to flash Merlin firmware. Flashed Asus Official 3.0.0.4.386.41535  first, worked fine. Flashed Merlin 386.1_0 next, worked fine. Then tried to flash 386.1_2 and got "Invalid Firmware Upload." Tried both the SourceForge and OneDrive downloads as well. I also tried flashing back to that Official firmware, which worked fine, then trying to flash 386.1_2 has the same error. 

**Model**
RT-AX86U

**Firmware version**
386.1_2

**Client information**
Windows 10, Chrome

**To Reproduce**
Be on 386.1_0 and try flashing 386.1_2

**Expected behavior**
Successfully flash the FW

**Actual results**
Invalid Firmware Upload

RMerl, 2021-02-15:
Works for me, please head to the support forum for assistance.

closed 2021-02-15 by RMerl

Strijd, 2021-06-06:
Hi 
Still persists, @WhiteZeroX did you manage to solve it ?

Strijd, 2021-06-07:
New firmware fixed this issue