RMerl/asuswrt-merlin.ng #702: [BUG] IGMP PROXY DOES NOT WORK

tommaso82, 2021-02-16:
/usr/sbin/igmpproxy it is not running even after activating it from the lan-iptv tab
trying to launch it manually fails using default cfg in /var/mcpd.conf

creating a working configuration I get this error

admin@RT-AC86U-1DE0:/tmp# /usr/sbin/igmpproxy /tmp/mcpd.conf -dv
MC-Router API already in use; Errno(98): Address already in use

test done on rt-ac86u with the latest merlin stable version

for those who want to use dlna servers across different subnets the only hope is to get igmpproxy to work

RMerl, 2021-02-24:
HND routers use mcpd, not igmpproxy.  That's why it's telling you it's already in use, probably because the mcpd daemon is already running.

closed 2021-02-24 by RMerl

tommaso82, 2021-03-10:
i have my merlin router (router mode) cascaded to my main isp-router on 192.168.1.0/24  (primary subnet), 192.168.2.0/24 is the subnet of the asus / merlin router 192.168.1.250 (eth0) is the ip of the merlin router in the subnet 192.168.1.0/24 , 192.168.2.1 (br0) is the ip of the asus / merlin in its network (wifi included). How can I make sure that the dlna servers present in the 192.168.1.x subnet can be visible to the clients of the asus / merlin router that are in the 192.168.2.x subnet? I tried editing mcpd.conf in every way, also tried to kill it and use igmpproxy but with no success