RMerl/asuswrt-merlin.ng #703: SSH from Router to Router Not Working

dbmajor, 2021-02-16:
Hi there,

My SSH from an RT-AC86U to an RT-AX88U and vice versa has stopped working.

Appears to have happened since 386.1_2 was released.

I can SSH from the laptop perfectly fine, but when I'm SSH'ing from router to router, it asks for a password instead of using the keys.

Could be a bug in the new release maybe?

Cheers
Josh

kolobock, 2021-02-17:
Hey, have you tried with -v flag? It will list all keys and methods used. After that, you will be able to see is the issue with pubkey missing or known_hosts record

dbmajor, 2021-02-17:
Thanks @kolobock,
I did try that, but it doesn’t seem to be a valid option when using ssh between routers I.e. ssh into or router 1 then ssh -v from router 1 to router 2.
The -v option does work when going from my computer to the router though.
Possibly a limitation of busybox?

RMerl, 2021-02-24:
Works for me, tested between an RT-AC68U and RT-AX88U.  However the server does not like comments, so when pasting your public key, remove the comment at the end, i.e. paste:

```
ssh-rsa AAABBBCD
```

And NOT

```
ssh-rsa ABBCCDD admin@myrouter
```

dbmajor, 2021-02-24:
> Works for me, tested between an RT-AC68U and RT-AX88U. However the server does not like comments, so when pasting your public key, remove the comment at the end, i.e. paste:
> 
> ```
> ssh-rsa AAABBBCD
> ```
> 
> And NOT
> 
> ```
> ssh-rsa ABBCCDD admin@myrouter
> ```

This worked for me, thanks @RMerl!

closed 2021-02-24 by dbmajor