RMerl/asuswrt-merlin.ng #704: Unable to update values on Administration=>System page

SergiiGalai, 2021-02-17:
**Describe the bug**
Administration => System page seems to be broken.
I've tried to enable the SSH access.
Logged in, opened System page, changed Service.SSH fields to enable SSH, pressed Apply.
Value preserved but the update popup did not appear. After page refresh settings restored to their previous state.
I have opened browser Developer Network tab to see requests.
Pressing on apply button does not send any request. Instead the page is scrolled up and displays the Router Login Name field with the name and error title 'This account already exists. Please enter a different name.'
It seems that settings page tries to send also user credentials to change user password with other fields even if the name was not updated. If Router Login Name field make empty then 'Username cannot be blank.' error appears

**Model**
RT-AC68U

**Firmware version**
386.1_2

**Client information**
Tried on Microsoft Edge 88.0.705.68, Firefox 85.0.2

**To Reproduce**
Login, go to Administration=>System
1. Change some value, e.g. set Enable SSH to another value than you have
2. Click apply
3. Observe selected user name and UI validation error 'This account already exists. Please enter a different name'
3. Refresh the page, see that Enable SSH restored its previous value

**Expected behavior**
System Apply button should update changed settings and user password should not be requested

**Actual results**
no request to server, page validation failed, request to change user name

**Will this error be reproduced on the official firmware?**

**Screenshots**
change values to update
[image]
press Apply
[image]
observe selected Router Login Name field
[image]
press F5, observe previous values
[image]

**Additional context**
Tried also to change other fields, the same result

hurda, 2021-02-17:
Does the router login name start with anything but a letter, or does the name contain anything but letters and numbers?

Have you ever imported settings from a different router/installation/version instead of applying the settings manually from scratch?

SergiiGalai, 2021-02-18:
In my case login name contains only lower case Latin letters.
No import. This is actually the first experience after flashing Merlin firmware on top of the stock firmware (the latest 386 version if that matters).

hurda, 2021-02-18:
I was asking because this user had the same issue, maybe the steps in posting #9 will work for you, too: https://www.snbforums.com/threads/fixed-cant-change-settings-on-administration-system-page-on-rt-ax88u-in-384-17.64582/

SergiiGalai, 2021-02-18:
Yes that was it, sorry for creating an issue. That looked to me as a bug.
Despite the installation with respect to https://github.com/RMerl/asuswrt-merlin.ng/wiki/Installation guideline which claims that factory is not mandatory. Thanks for help! Maybe it worth to close the issue but leave it searchable so other guys could not spend a lot of time and disturb you as well...

closed 2021-02-18 by SergiiGalai