RMerl/asuswrt-merlin.ng #705: DNS over TLS config forces Round-Robin recursive resolvers

plmcgrn, 2021-02-19:
In looking at the text of `stubby.yml ` that configures the DNS over TLS support, I found that this line seems to hard code round robin for upstream DNS.  Since there's no UI element to configure this behavior, I assume it'd be a source code edit or some edit to the file system on my router.  I haven't looked deeply into how those work, but I know it's not as simple as editing stubby.yml since that won't persist due to being loaded into /tmp.

`round_robin_upstreams: 1`

This round robin causes issues when a user may want the DNS servers to be queried top-down, instead of round robin.  Is there a way to modify this setting persistently?

dave14305, 2021-02-20:
Look in the wiki for “User Scripts” to setup /jffs/scripts/stubby.postconf to replace the existing line with your preferred line.

plmcgrn, 2021-02-20:
Thanks, I will look into that.  From a quick skim, it looks like I create a bash script called stubby.postconf in /jffs/scripts which runs pc_replace to replace that relevant line in the existing stubby.yml config file.

Before I get too deep into that, I want to test if re-setting the value via stubby.yml.add will actually achieve the same behavior (assuming stubby reads conf top to bottom and initializes afterwards), without a boot-blocking script.

plmcgrn, 2021-02-20:
Closing issue, as workaround is workable, and I agree in principle that round robin is an OK default for most scenarios.

closed 2021-02-20 by plmcgrn