RMerl/asuswrt-merlin.ng #708: mDNS does not appear to be working since 386.1

stevesch, 2021-02-27:
mDNS devices on LAN normally accessible via http://<name>.local/ are no longer accessible.
Tried on 386.1, 386.1_2.  Rebooting, etc., did not resolve issue.
Access by direct IP (192.68.x.x) appears fine.  Devices tried include several ESP32-based boards, Logitech hubs.
If this a known issue or there are setting changes that could affect this, info would be appreciated.

dave14305, 2021-02-27:
I was just able to access my printer web interface at http://hpoj8600.local/ on 386.1_2 on an RT-AC86U.

kolobock, 2021-02-27:
Most of  `ESP32-based boards` are working in energy-saving mode so can be
available on the net periodically.

RMerl, 2021-02-27:
A much better test would be to test name resolution itself rather than testing client access with a browser.

stevesch, 2021-02-28:
nslookup fails for these devices as well-- though, I didn't test them using this prior to my current issue.
The mDNS implementation on the devices may not be compliant, so I'm not sure it's really a router issue, but I'm trying to track down how to make them function again.  Thanks for the info

stevesch, 2021-03-02:
I fixed/worked-around my issue by setting
LAN > DHCP Server > RT-AX88U's Domain Name
to "local"
Is this the proper fix, i.e. is it expected that mDNS ".local" addresses require this in order to be found? (the default value is blank, correct?).

Non-working behavior: for wireless device "foo" on LAN
- can be accessed by all LAN clients via http://192.168.0.172/
- can be accessed by other WIRELESS LAN clients via http://foo/
- CANNOT be accessed by wired LAN clients via http://foo/
- CANNOT be accessed by any LAN clients via http://foo.local/

Working behavior, after setting "RT-AX88U's Domain Name" to "local":
- can be accessed by all LAN clients via http://192.168.0.172/
- can be accessed by all LAN clients via http://foo.local/
- can be accessed by all LAN clients via http://foo/

RMerl, 2021-03-08:
> I fixed/worked-around my issue by setting
> LAN > DHCP Server > RT-AX88U's Domain Name
> to "local"
> Is this the proper fix, i.e. is it expected that mDNS ".local" addresses require this in order to be found? (the default value is blank, correct?).

No.  What you just did was replace mDNS with regular DNS.

I just tested mDNS on my own LAN, and it seems to work fine for me on my network with an RT-AX88:

```
merlin@ubuntu-dev:~$ dig camelot.local @224.0.0.251 -p 5353 +short
192.168.10.100
merlin@ubuntu-dev:~$ dig alexandria.local @224.0.0.251 -p 5353 +short
192.168.10.200
```

stevesch, 2021-03-11:
I believe you're correct that the router is functioning properly.  At most, something is behaving a little differently since <386.x, but I think my mDNS connectivity issues are due to something else happening my network.
I have yet to resolve the exact issue, but thanks for the time and tests.  I'll close this issue.

closed 2021-03-11 by stevesch

stevesch, 2021-04-08:
For anyone encountering this: my issue was solved be enabling "IGMP Snooping" (for both the 2.4GHz and 5GHz radios).
Wireless > Professional > 2.4 GHz :: Enable IGMP Snooping = Enable (was Disable)
Wireless > Professional > 5 GHz :: Enable IGMP Snooping = Enable (was Disable)
(domain name mentioned in previous post can be left blank)

I'm not certain why my mDNS devices worked prior to the update, at least intermittently, but with IGMP snooping enabled the devices now connect reliably.