RMerl/asuswrt-merlin.ng #713: Question about local NTP server and DST change in Europe

Trainax, 2021-03-04:
Hello, I didn't use the template because this is not a proper bug.

In Europe DST time starts on the **last** Sunday of March which is not always the 4th or the 5th week as statically set in the menu. Same thing applies for DST time ending which is on the **last** Sunday of October (which could be the 4th or the 5th). Would it be possible to have an option to select "Last" from the drop down menu instead of "4th" or "5th" so the system reflects the correct behaviour without having to manually change the settings every year?

[image]

Firmware Version : 386.1_2

Thank you for your awesome work!

maghuro, 2021-03-06:
5th also means last. Use it, don't need to change it every year

Trainax, 2021-03-07:
> 5th also means last. Use it, don't need to change it every year

So if I select "5th" and the 5th Sunday of the month does not exist the time changes on the 4th Sunday automatically without my intervention?

maghuro, 2021-03-07:
It won't changes, the interface keeps the 5th, but it'll always assume the last Sunday of month.
Trust me :)

Trainax, 2021-03-07:
> 
> 
> It won't changes, the interface keeps the 5th, but it'll always assume the last Sunday of month.
> Trust me :)

Ok. Thank you for your reply!

I'm closing this issue as my question is solved :-)

Trainax, 2021-03-18:
I'm reopening this because even if I set the DST change to happen on the 5th Sunday of the month the router time still changed last Sunday (2nd Sunday of the month) following American standard.

[image]

Now the router time is one hour ahead even if the Time Zone inside is correct (I live in Italy). As you can see DST is enabled even if we didn't yet make the change in my country.

[image]

DST will start on the last Sunday of this month (Sunday 28th of March)

snaunton, 2021-03-18:
"* Daylight savings time is enabled in this time zone." does not mean that the time is currently DST,  it just means that this timezone has DST.

When I set my router (RT-AC68U) to the same timezone as you, if I check the time against duckduckgo-ing "what is the time in rome" the time is correct.

# cat /etc/TZ
MEZ-1DST,M3.5.0/2,M10.5.0/3

Trainax, 2021-03-18:
> When I set my router (RT-AC68U) to the same timezone as you, if I check the time against duckduckgo-ing "what is the time in rome" the time is correct.

Thank you for testing, however I can confirm it's 21:57 now in Italy but the router time is 22:57.

Router time was correct until last Sunday when it jumped forward

snaunton, 2021-03-18:
I was wrong.

release/src/router/www/Main_LogStatus_Content.asp:

`function showDST(){
	var system_timezone_dut = "<% nvram_get("time_zone"); %>";
	if(system_timezone_dut.search("DST") >= 0 && "<% nvram_get("time_zone_dst"); %>" == "1"){
		document.getElementById('dstzone').style.display = "";
		document.getElementById('dstzone').innerHTML = "<#General_x_SystemTime_dst#>";
	}
}`

Can you log into the router via SSH and execute:

nvram show | grep -i ^time_zone

And post the output?

snaunton, 2021-03-18:
Can you also ssh into the router, execute the following and post it here?

cat /etc/TZ

snaunton, 2021-03-18:
Actually, ignore all my previous posts. I seem to be correct in my initial post that "* Daylight savings time is enabled in this time zone." does not mean that the time is currently DST, it just means that this timezone has DST.

The nvram value time_zone_dst only means that a timezone has DST.

What is the output of:

cat /etc/TZ

It is this file that controls what the actual timezone settings are for the OS.

Trainax, 2021-03-19:
> What is the output of:
> cat /etc/TZ

The output is:
MEZ-1DST

snaunton, 2021-03-19:
This is the same bug that my fix below fixes (same problem with GMT+10 Canberra, Melbourne, Sydney:

https://github.com/RMerl/asuswrt-merlin.ng/pull/701

I have thie fix running on my router which is why I couldn't see the problem.

This is in 386.2-beta1, so I assume would be in 386.2 when released.

In the mean time ssh in and execute the following as a workaround.

echo "MEZ-1DST,$(nvram get time_zone_dstoff)" > /etc/TZ

If this doesn't work (I am guessing somewhat), please post:

cat /etc/TZ
nvram show | grep ^time_zone

You would have to do this after every reboot.

Trainax, 2021-03-19:
I have executed the first command however it didn't fix the issue (it's 9:45 but the router says it's 10:45).

Here is the output of the commands

[image]

snaunton, 2021-03-19:
Try:

nvram set time_zone_x=MEZ-1DST,M3.5.0/2,M10.5.0/3
nvram get time_zone_x > /etc/TZ

Trainax, 2021-03-19:
> Try:
> 
> nvram set time_zone_x=MEZ-1DST,M3.5.0/2,M10.5.0/3
> nvram get time_zone_x > /etc/TZ

Thank you! It worked! The time on the router is correct now.

Do you think this fix will be included in 386.2 so I don't have to do this every time I reboot the router?

snaunton, 2021-03-19:
The fix should be included.

Remember that you have to run those two commands if you reboot the router.

RMerl, 2021-03-27:
Closing, as this is related to something already fixed in 386.2 beta.

closed 2021-03-27 by RMerl