RMerl/asuswrt-merlin.ng #720: Changing reboot schedule forces new Router Login Name and Password

FastThenLeft, 2021-03-15:
**Describe the bug**
On the Administration->System page: To change the reboot schedule, you are forced to change the Router Login Name and Password. If you leave them blank and only change the schedule, it says "Username Cannot Be Blank".

**Model**
RT-AC5300

**Firmware version**
386.1_2

**Client information**
Chrome Version 89.0.4389.90 (Official Build) (64-bit)

**To Reproduce**
Go to the Administration->System page and try and change the reboot schedule without changing the Router Login Name and Password.

**Expected behavior**
I would expect to be able to change anything on the Administration->System page without also having to change the Router Login Name and Password.

**Actual results**
I am forced to also change the Router Login Name and Password just to change the reboot schedule.

RMerl, 2021-03-17:
The username field isn't supposed to be blank, it should contain your current username.  Only password fields can be blanks, as they are only used when changing password.  Are you saying your Username field is empty when you access the page?

FastThenLeft, 2021-03-17:
If I put in my current username in "Router Login Name" and only try to
change the schedule, it says "This account already exists. Please enter a
different name."

If I leave it blank, it says, "Username cannot be blank."

There appears no way to change the schedule without changing both the
username and password.

RMerl, 2021-03-17:
That doesn't answer the question...  Is the username field empty when you load the page?  If yes, then the issue is something is preventing the page from fully loading.  Open your browser console and look for any Javascript error at load time.  Please post the content of the line where the Javascript error happens.

FastThenLeft, 2021-03-17:
No. The username is filled in when the page loads.

RMerl, 2021-03-17:
I don't see how it could complain about it being empty then.  I checked the page code, and that error message only shows if the length of the username is 0.

```
        if(document.form.http_username.value.length == 0){
                showtext(document.getElementById("alert_msg1"), "<#File_Pop_content_alert_desc1#>");
                document.form.http_username.focus();
                document.form.http_username.select();
                return false;
        }
```
Have you tried either disabling browser plugins or Incognito mode to ensure it's not something in the browser interfering?

FastThenLeft, 2021-03-17:
Sorry if I confused you a bit. When I go to the page, the username is
already filled in. Then I go to the schedule and change it. Then I go to
the bottom of the page and click, "Apply" and it jumps me back up to the
top of the page and under the "Router Login Name" it says, "This account
already exists. Please enter a different name."

I've tried Incognito mode and it works the same way. I also tried it with a
RAM based Linux so that browser didn't have any plugins and it worked the
same.

RMerl, 2021-03-17:
Gotcha.  This is a different issue then, it's related to USB disk sharing accounts getting out of sync with the router's main admin account.  The issue appeared a few months ago (possibly with the move to 386_xx), however I haven't been able to fully track it down yet.

FastThenLeft, 2021-03-17:
OK, please let me know if I can help in any way. Thanks so much for making
such great software!

RMerl, 2021-03-18:
Can you post the result from this command?

```
nvram get acc_list
```

Feel free to replace the random portion with "password", as this is a hashed password.  I mostly need to see the content and the format of that variable on your system.  It sounds like the first account in that list is not the router admin account, which is causing the error message you are encountering.

FastThenLeft, 2021-03-18:
david>+password+password==<MediaMachine>password==
I hope I didn't replace too much with "password".

RMerl, 2021-03-18:
Are either "david" or "MediaMachine" your router's admin username?

The router expect the first account in acc_list to be identical to the router's admin user.  If they are in the wrong order, you might have to swap them (along with their passwords).

FastThenLeft, 2021-03-18:
Yes, MediaMachine. Does that mean it's the second one and how do I swap it?
Or, get rid of "david"?

FastThenLeft, 2021-03-18:
If "nvram get acc_list" produced:
david>+password1+password2==<MediaMachine>password3==

Do I do:
nvram set acc_list="MediaMachine>password3==<david>+password1+password2=="
To swap them?

RMerl, 2021-03-18:
Try first just removing "david" from the list of USB users, then re-add it.  Check if it added it at the end as expected.

Otherwise yes, you could manually swap them using nvram set acc_list="" (also run "nvram commit" afterward to save changes to flash).

FastThenLeft, 2021-03-18:
I first removed "MediaMachine from the list of USB users, and it fixed it!
nvram get acc_list
david>+password+password==

RMerl, 2021-03-18:
You should now be able to make changes to the System page without the error message.

FastThenLeft, 2021-03-18:
Yes, that fixed it. Thanks!

closed 2021-03-20 by RMerl