RMerl/asuswrt-merlin.ng #723: Parent Folder Missing in Samba share for RT-AX86U on 386.1_2

thunder2k, 2021-03-19:
Hi,
I have two USB3.0 drives connected to my router.

If I enable samba share, all the folders in both drives got mixed in a single list. In particular, I have the same structure in both HDD so some folders are not being accessible.

If I disable Simpler share naming (without the disk name) I can clearly see all folders with the parent folder name in brackets.

I would have expected a parent folder matching the one in /mnt in the router containing all the subfolders.

Am I missing something here?

Can I achieve this modifying some settings?

Thanks!

Bye

RMerl, 2021-03-20:
Working as intended - the root cannot be shared, only folders.

closed 2021-03-20 by RMerl

thunder2k, 2021-03-20:
> Working as intended - the root cannot be shared, only folders.

Hi,
Why cannot be shared? With SMB1 and no authentication it works this way.

I ended up overriding the config in JFFS partition and defining the samba share exactly like that.

RMerl, 2021-03-20:
Config data is stored at the root of the disk, so that would be a major security risk.

thunder2k, 2021-03-20:
I see. Can you not save it in the JFFS partition? Is that to minimize the JFFS I/O and reduce wearing?

I assume my config is safe enough, isn't it?

Also, if I reboot the router without one or both HDDs attached, would my override config keep the router from booting properly? I've copied the smb.conf file from the router and modified just the folders bit.

RMerl, 2021-03-20:
Asus designed it so the user data is stored at the root of the drive, so each drive can have its own data, and it will be kept in sync/persistent across devices or if a disk is reformated.

Just create a sub-folder and store all your files there.  Every single NAS also work the same way, none of them will share the root of the disk, only sub-folders.  It's accepted best-practice, and totally transparent to end-users.

thunder2k, 2021-03-20:
Makes total sense.

I was lazy to add the root folder in the HDD since I have a lot of stuff pointing to that folder structure, but I guess I could just change the mount point :)

Thanks a lot for taking the time to educate me.

I flashed your firmware as soon as I unboxed the router, the per-device VPN client setup got me (among others) :)

Keep up the good work! 

Have a nice day!

rhjdvsgsgks, 2021-05-22:
you can have a try with this little script
https://github.com/rhjdvsgsgks/MerlinSambaLocationConfig/blob/master/sambalocationconfig.sh
put it to your `smb.postconf` , and add your "location config" to `/mnt/sda1/sambalocationconfig.conf`
such as
```
[sda1]
veto files = /.__*.txt*/asusware*/asus_lighttpdpasswd/
path = /tmp/mnt/sda1
dos filetimes = yes
fake directory create times = yes
valid users = admin, aaa
invalid users = aaa
read list = admin
write list = admin
```
and now you can access parent folder "sda1"