RMerl/asuswrt-merlin.ng #724: [AX88U][386.1/2]"service restart_net" break the entire WIFI if 1st slot Guest Network is enabled.

kissglay, 2021-03-21:
**Describe the bug**

If the 1st slot of Guest network is ALREADY enabled BEFORE booting into the router, executing `service restart_net` (disabling and then enabling IPV6 from WebUI can achieve the same result)  breaks the entire WIFI. Both WIFI bands are still available, but no devices can connect to them. Following entries are filling the log:
`hostapd: eth6: STA 40:31:3c:a9:bb:af IEEE 802.11: disassociated
hostapd: eth6: STA 7c:49:eb:8d:ab:58 IEEE 802.11: associated
kernel: CFG80211-ERROR) wl_cfg80211_change_station : WLC_SCB_AUTHORIZE sta_flags_mask not set 
hostapd: eth6: STA 34:ce:00:fd:72:ec IEEE 802.11: associated
kernel: CFG80211-ERROR) wl_cfg80211_change_station : WLC_SCB_AUTHORIZE sta_flags_mask not set 
hostapd: eth6: STA 04:cf:8c:1b:f1:f5 IEEE 802.11: disassociated
hostapd: eth6: STA 38:d2:ca:d5:1c:26 IEEE 802.11: associated
kernel: CFG80211-ERROR) wl_cfg80211_change_station : WLC_SCB_AUTHORIZE sta_flags_mask not set 
hostapd: eth6: STA 7c:49:eb:8d:ab:58 IEEE 802.11: disassociated
hostapd: eth6: STA 34:ce:00:fd:72:ec IEEE 802.11: disassociated
hostapd: eth6: STA 04:cf:8c:1b:f1:f5 IEEE 802.11: associated
kernel: CFG80211-ERROR) wl_cfg80211_change_station : WLC_SCB_AUTHORIZE sta_flags_mask not set 
hostapd: eth6: STA 40:31:3c:a9:bb:af IEEE 802.11: associated
kernel: CFG80211-ERROR) wl_cfg80211_change_station : WLC_SCB_AUTHORIZE sta_flags_mask not set 
hostapd: eth6: STA 38:d2:ca:d5:1c:26 IEEE 802.11: disassociated
hostapd: eth6: STA 04:cf:8c:0b:d8:87 IEEE 802.11: associated
kernel: CFG80211-ERROR) wl_cfg80211_change_station : WLC_SCB_AUTHORIZE sta_flags_mask not set`

Note that if you have Guest WIFI off, boot into the router and then enable them, or you set up the Guest Network in 2nd slot (not tested in 3rd), this bug no longer stands.

**Model**

AX88U

**Firmware version**

From the very first alpha of 386.1 and forth

**Client information**

N/A

**To Reproduce**

1. Turn on the Guest Network in 1st slot
2. Reboot the router.
3. Execute `service restart_net`

**Expected behavior**

Both WIFI bands should be normal.

**Actual results**

Both bands are still there, but nothing can connect to them. 

**Will this error be reproduced on the official firmware?**

Not tested

**Screenshots**

**Additional context**

paldier, 2021-03-22:
Confirm whether the eapd process exists, if it does not exist, execute /bin/eapd &

kissglay, 2021-03-22:
Thx for the tip. 
But, yes, it exists in both cases I mentioned.
So it's not the cause.

RMerl, 2021-03-27:
There are all kind of new issues introduced with Asus's new Guest Network implementation used by the first one.  I'm leaving it to them to sort these things out.