RMerl/asuswrt-merlin.ng #726: Wrong timezone parsing in frontend

gabortim, 2021-03-23:
**Describe the bug**

Some browser locales break the timezone parser lying in the state.js file, [here](https://github.com/RMerl/asuswrt-merlin.ng/blob/f0b75788872a2e05c40a7fd086caf9aedb8cf43b/release/src/router/www/state.js#L3621), resulting in a misleading message at `Advanced Settings | Administration | System | Time Zone`.

**Model**

RT-AX58U, but probably most current and legacy AsusWRT based routers affected.

**Firmware version**

Merlin 386.1_2

**Client information**

Google Chrome 89.0.4389.90 x64 with Hungarian locale
Mozilla Firefox 86.0.1 x64 with Hungarian locale

**To Reproduce**

Set browser locale to a language where the JavaScript `new Date()` constructor returns a string with an extra plus or minus sign in it.
Example with Hungarian locale: `Tue Mar 23 2021 14:58:29 GMT+0100 (közép-európai téli idő)`.

The parser first searches for the last occurrence of a `-` or `+` char in the returned string and then compares to the stored value (eg. `+0100`). In this case `-euró != +0100`, so the message appears.

**Expected behavior**

No warning about time zone shift.

**Actual results**

_\* Reminder: The System time zone is different from your locale setting._

**Will this error be reproduced on the official firmware?**

Yes, it's a long-standing issue.

**Screenshots**

[image]

**Additional context**

The issue is already reported to Asus in February but got only machine answer.

Trainax, 2021-03-24:
I had a similar problem, take a look at #713 and see if your problem is the same as mine and if the solution which was given to me also works for you.

A fix for this should come with version 386.2

snaunton, 2021-03-24:
@Trainax this is a different problem. I am running the fix for @Trainax problem and I still get the spurious "* Reminder: The System time zone is different from your locale setting."

I just ignore it. It doesn't impact any functionality.

gabortim, 2021-03-24:
Yep, the mentioned one is a different issue. I have a fix for this, but I don't know what are the browser compatibility requirements.

That's true, it doesn't impact any functionality but already triggered so many comments on the internet, it worth fixing.

Trainax, 2021-03-24:
> 
> 
> Yep, the mentioned one is a different issue. I have a fix for this, but I don't know what are the browser compatibility requirements.
> 
> That's true, it doesn't impact any functionality but already triggered so many comments on the internet, it worth fixing.

I misunderstood your problem. Sorry

snaunton, 2021-03-27:
What is the point of the code? What does it matter if the browser timezone is different from the timezone of the router?

snaunton, 2021-03-27:
@gabortim - noticed the confused smiley, and assuming that you are confused by my post. 

The code takes (well it would if it wasn't buggy) the timezone of the browser and then compares it to the timezone of the router. If they are different it shows the message.

Who cares about this?

gabortim, 2021-03-27:
As I wrote previously, I have read many forum comments on the internet about this particular warning because people got confused. Rightly so, because everything is set up correctly and it's still there.

There are two options:
- disable the warning altogether
- fix it

I opt for the second option but I'm curious what's your choice. I have a fix for this but also an impression you didn't want to see it fixed because you do not see it as an actual issue 😕

commit 73e5ec95 (2021-03-27): webui: Fix timezone detection (#729)

Ignore text in parentheses which may include a dash in non-English languages and confuse the index logic.  For example:

"GMT+0300 (Itä-Euroopan kesäaika)"

Fixes #726
closed 2021-03-27 by RMerl