RMerl/asuswrt-merlin.ng #728: RT-AX56U && RT-AX58U allowing privileges to properly drop

jumpsmm7, 2021-03-24:
**Describe the bug**

There is an issue exclusive to RT-AX56U && RT-AX58U dropping privileges running Dnscrypt-proxy 2 which suggest something with the kernal build.  

**Model**

RT-AX56U && RT-AX58U 

**Firmware version**

386.2 Beta 2

**Client information**

**To Reproduce**

Install dnscrypt-proxy 2 on RT-AX56U && RT-AX58U using install script for asuswrt-merlin.

**Expected behavior**

 privileges to behave properly

**Actual results**

privileges do not properly function for these models to run dnscrypt-proxy 2

**Will this error be reproduced on the official firmware?**

N/A

**Screenshots**

**Additional context**

RMerl, 2021-03-24:
CONFIG_MULTIUSER=y is already present, so the issue lies elsewhere.

```
admin@RT-AX58U-E588:/tmp/home/root# zcat /proc/config.gz | grep CONFIG_MULTIUSER
CONFIG_MULTIUSER=y
```
The same issue can be observed with dnsmasq, where both threads run as admin instead of having one drop to nobody.

RMerl, 2021-03-24:
Correction - dnsmasq does drop privileges properly on the RT-AX58U.  So this looks more like a bug with dnscrypt and not the firmware.

RMerl, 2021-03-25:
I also confirmed that privileges dropping worked fine, with the following test program:

```
#include <stdio.h>
#include <sys/types.h>
#include <unistd.h>
#include <grp.h>

int main(){
	printf("start\n");
	printf("sleep\n"); usleep(5 * 1000 * 1000);

	uid_t uid = 65534; // user
	gid_t gid = 65534;  // group

        printf("setgid: %d\n", uid);
        printf("%d\n",setgid(uid));

        printf("setuid: %d\n", gid);
        printf("%d\n",setuid(gid));

	usleep(20*1000*1000);
	printf("exit\n");
}
```

If I examine the process during that 20 secs pause, it does show gid and uid of 65534 (both of which are nobody), and both calls return "0".

jumpsmm7, 2021-03-25:
> 
> 
> I also confirmed that privileges dropping worked fine, with the following test program:
> 
> ```
> #include <stdio.h>
> #include <sys/types.h>
> #include <unistd.h>
> #include <grp.h>
> 
> int main(){
> 	printf("start\n");
> 	printf("sleep\n"); usleep(5 * 1000 * 1000);
> 
> 	uid_t uid = 65534; // user
> 	gid_t gid = 65534;  // group
> 
>         printf("setgroups()\n");
>         printf("%d\n",setgroups(0, NULL));
>         printf("setgid: %d\n", gid);
>         printf("%d\n",setgid(gid));
> 
>         printf("setuid: %d\n", uid);
>         printf("%d\n",setuid(uid));
> 
> 	usleep(20*1000*1000);
> 	printf("exit\n");
> }
> ```
> 
> If I examine the process during that 20 secs pause, it does show gid and uid of 65534 (both of which are nobody), and both calls return "0".

So in theory, Dnscrypt-proxy2 should be working than. The branch I am using is Linux-arm. The router uses `armv7l`. So I would assume there is no way I could be using the wrong build if that works with all the other `armv7l` run models?

RMerl, 2021-03-27:
Closing this as ultimately it's a Go issue, not a firmware issue.

closed 2021-03-27 by RMerl