RMerl/asuswrt-merlin.ng #733: OpenVPN client stop works after alest update

panther7, 2021-04-02:
**Describe the bug**
OpenVPN client stop works after last update, v386.1_2 works properly.

**Model**
RT-AC66U (B1)

**Firmware version**
386.2

**Actual results**
```
Apr  2 17:57:24 rc_service: httpds 866:notify_rc start_vpnclient1
Apr  2 17:57:26 ovpn-client1[7788]: WARNING: Compression for receiving enabled. Compression has been used in the past to break encryption. Sent packets are not compressed unless "allow-compression yes" is also set.
Apr  2 17:57:26 ovpn-client1[7788]: Note: option tun-ipv6 is ignored because modern operating systems do not need special IPv6 tun handling anymore.
Apr  2 17:57:26 ovpn-client1[7788]: --cipher is not set. Previous OpenVPN version defaulted to BF-CBC as fallback when cipher negotiation failed in this case. If you need this fallback please add '--data-ciphers-fallback BF-CBC' to your configuration and/or add BF-CBC to --data-ciphers.
Apr  2 17:57:26 ovpn-client1[7788]: OpenVPN 2.5.1 arm-unknown-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] built on Apr  1 2021
Apr  2 17:57:26 ovpn-client1[7788]: library versions: OpenSSL 1.1.1k  25 Mar 2021, LZO 2.08
Apr  2 17:57:26 ovpn-client1[7789]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Apr  2 17:57:26 ovpn-client1[7789]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Apr  2 17:57:26 ovpn-client1[7789]: Outgoing Control Channel Authentication: Using 160 bit message hash 'SHA1' for HMAC authentication
Apr  2 17:57:26 ovpn-client1[7789]: Incoming Control Channel Authentication: Using 160 bit message hash 'SHA1' for HMAC authentication
Apr  2 17:57:26 ovpn-client1[7789]: TCP/UDP: Preserving recently used remote address: [AF_INET]77.75.74.222:1193
Apr  2 17:57:26 ovpn-client1[7789]: Socket Buffers: R=[122880->122880] S=[122880->122880]
Apr  2 17:57:26 ovpn-client1[7789]: UDP link local: (not bound)
Apr  2 17:57:26 ovpn-client1[7789]: UDP link remote: [AF_INET]77.75.74.222:1193
Apr  2 17:57:26 ovpn-client1[7789]: TLS: Initial packet from [AF_INET]77.75.74.222:1193, sid=7ead8f26 6615cf28
Apr  2 17:57:26 ovpn-client1[7789]: VERIFY OK: depth=1, C=CZ, ST=Czech Republic, L=Prague, O=Seznam.cz, a.s., CN=Seznam.cz Kancelar VPN Root
Apr  2 17:57:26 ovpn-client1[7789]: VERIFY OK: depth=0, C=CZ, ST=Czech Republic, O=Seznam.cz, a.s., CN=vpn-server-kancelar
Apr  2 17:57:26 ovpn-client1[7789]: WARNING: 'link-mtu' is used inconsistently, local='link-mtu 1542', remote='link-mtu 1558'
Apr  2 17:57:26 ovpn-client1[7789]: WARNING: 'keysize' is used inconsistently, local='keysize 128', remote='keysize 256'
Apr  2 17:57:26 ovpn-client1[7789]: Control Channel: TLSv1.3, cipher TLSv1.3 TLS_AES_256_GCM_SHA384, 2048 bit RSA
Apr  2 17:57:26 ovpn-client1[7789]: [vpn-server-kancelar] Peer Connection Initiated with [AF_INET]77.75.74.222:1193
Apr  2 17:57:28 ovpn-client1[7789]: SENT CONTROL [vpn-server-kancelar]: 'PUSH_REQUEST' (status=1)
Apr  2 17:57:28 ovpn-client1[7789]: PUSH: Received control message: 'PUSH_REPLY,tun-ipv6,route-ipv6 2a02:598::/32,topology subnet,route-gateway 10.0.112.1,route 10.0.0.0 255.255.0.0,route 10.8.0.0 255.255.0.0,route 10.4.0.0 255.255.0.0,route 10.3.0.0 255.255.0.0,route 10.2.0.0 255.255.0.0,route 10.1.0.0 255.255.0.0,route 10.233.0.0 255.255.0.0,route 10.234.0.0 255.255.0.0,route 10.235.0.0 255.255.0.0,route 10.236.0.0 255.255.0.0,route 10.237.0.0 255.255.0.0,route 10.238.0.0 255.255.0.0,route 10.239.0.0
Apr  2 17:57:28 ovpn-client1[7789]: PUSH: Received control message: 'PUSH_REPLY,route 10.255.255.0 255.255.255.0,route 10.255.7.0 255.255.255.0,route 10.16.0.0 255.240.0.0,route 10.32.0.0 255.255.0.0,route 10.46.0.0 255.255.0.0,route 10.48.0.0 255.240.0.0,route 10.64.0.0 255.224.0.0,route 10.96.0.0 255.224.0.0,route 10.128.0.0 255.224.0.0,route 10.100.0.0 255.255.0.0,route 10.160.0.0 255.255.0.0,route 80.79.31.26 255.255.255.255,route 82.142.116.113 255.255.255.255,route 77.75.72.0 255.255.255.0,route 
Apr  2 17:57:28 ovpn-client1[7789]: PUSH: Received control message: 'PUSH_REPLY,route 77.75.79.0 255.255.255.0,route 77.75.79.155 255.255.255.255 net_gateway,route 10.250.20.222 255.255.255.255,route 10.250.50.222 255.255.255.255,route 10.255.0.172 255.255.255.255,route 10.0.2.130 255.255.255.255,route 10.0.3.46 255.255.255.255,route 90.176.150.207 255.255.255.255,dhcp-option DNS 10.0.2.124,dhcp-option DNS 10.0.2.125,dhcp-option DOMAIN ko.seznam.cz,dhcp-option DOMAIN ng.seznam.cz,dhcp-option DOMAIN oa.
Apr  2 17:57:28 ovpn-client1[7789]: OPTIONS IMPORT: timers and/or timeouts modified
Apr  2 17:57:28 ovpn-client1[7789]: OPTIONS IMPORT: --ifconfig/up options modified
Apr  2 17:57:28 ovpn-client1[7789]: OPTIONS IMPORT: route options modified
Apr  2 17:57:28 ovpn-client1[7789]: OPTIONS IMPORT: route-related options modified
Apr  2 17:57:28 ovpn-client1[7789]: OPTIONS IMPORT: --ip-win32 and/or --dhcp-option options modified
Apr  2 17:57:28 ovpn-client1[7789]: OPTIONS IMPORT: peer-id set
Apr  2 17:57:28 ovpn-client1[7789]: OPTIONS IMPORT: adjusting link_mtu to 1625
Apr  2 17:57:28 ovpn-client1[7789]: OPTIONS IMPORT: data channel crypto options modified
Apr  2 17:57:28 ovpn-client1[7789]: Data Channel: using negotiated cipher 'AES-256-GCM'
Apr  2 17:57:28 ovpn-client1[7789]: Outgoing Data Channel: Cipher 'AES-256-GCM' initialized with 256 bit key
Apr  2 17:57:28 ovpn-client1[7789]: Incoming Data Channel: Cipher 'AES-256-GCM' initialized with 256 bit key
Apr  2 17:57:28 ovpn-client1[7789]: GDG6: remote_host_ipv6=n/a
Apr  2 17:57:28 ovpn-client1[7789]: net_route_v6_best_gw query: dst ::
Apr  2 17:57:28 ovpn-client1[7789]: net_route_v6_best_gw result: via :: dev lo
Apr  2 17:57:28 ovpn-client1[7789]: TUN/TAP device tun11 opened
Apr  2 17:57:28 ovpn-client1[7789]: TUN/TAP TX queue length set to 1000
Apr  2 17:57:28 ovpn-client1[7789]: /usr/sbin/ip link set dev tun11 up mtu 1500
Apr  2 17:57:28 ovpn-client1[7789]: /usr/sbin/ip link set dev tun11 up
Apr  2 17:57:28 ovpn-client1[7789]: /usr/sbin/ip addr add dev tun11 10.0.117.216/20
Apr  2 17:57:28 ovpn-client1[7789]: Linux ip addr add failed: external program exited with error status: 2
Apr  2 17:57:28 ovpn-client1[7789]: Exiting due to fatal error
```

panther7, 2021-04-02:
Solved

closed 2021-04-02 by panther7

hurda, 2021-04-03:
And how?

panther7, 2021-04-03:
> And how?

VPN servers was upgraded. In router, I had to do enable IPv6.