RMerl/asuswrt-merlin.ng #735: New wiki entry: Compile-Firmware-from-source (multi-platform)

1951FDG, 2021-04-07:
Hi @RMerl,

I've recently had to compile the firmware to compile additional kernel modules not present and I've written a definitive guide on how to compile the firmware on multiple platforms using [Multipass](https://multipass.run):

[Compile-Firmware-from-source](https://github.com/RMerl/asuswrt-merlin.ng/wiki/Compile-Firmware-from-source)

If you  could revise it quickly, especially the part:

Note: Do not build from the master branch, build from the mainline.384 branch or ax.384 branch.

I am not entirely sure if one has to use the master branch, or otherwise. Hopefully I got the above note right in the wiki...

RMerl, 2021-04-08:
The 384.xxxx branches are over 6 months old.  The current code is on master at this time.

1951FDG, 2021-04-08:
Ok, I've made the changes to wiki entry! Thanks for the info.

closed 2021-04-08 by 1951FDG