RMerl/asuswrt-merlin.ng #738: LAN ports 5-8 appear disconnected

diabliya, 2021-04-08:
Hello, 

`Description: The ports 5-8 (LAN) appear disconnected when there are devices connected to them.`

- Model router: Asus AC88U
- Current firmware: 386.2
- Old firmware: 384.19_0 (in this firmware the problem wasn't exist)

**Expected behaviour**: It should appears the speed of the device connected, like it is in the others ports. 

**_Additional information_**: I was impacted in the past for this problem but it was resolved in the 384.19 version I guess (I am not 100% sure if was this asuswrt version but If I am not wrong, the problem was solved before I upgrade to 386 version and the last version I had was 384.19).

At the begining I was resigned to be impacted for the electrical issue as people explain in some forums but when the issue was fixed in some version of asuswrt, I though that maybe the real problem wasn't electrical but the workaround was electrical reset.

Thanks.

RMerl, 2021-04-08:
Have you tried power cycling the router?

diabliya, 2021-04-08:
Hello, 

Thanks for your quick response RMerl.
I was doing some testing and the following the next steps:

1. Installed the firmware 314.19 again in the router 
2. Then I returned to install the last version 316.2.

Now it is working the 8 ports with a simple restart.

closed 2021-04-10 by diabliya