RMerl/asuswrt-merlin.ng #739: ax11000 368.2 version ddns problem

suizhenmin, 2021-04-10:
The WWW.TUNNELBROKER.NET ddns service in 368.2 firmware of gt-ax11000 cannot be verified normally.
Below is the error message.
Apr  8 14:44:50 kernel: CFG80211-ERROR) wl_cfg80211_change_station : WLC_SCB_AUTHORIZE sta_flags_mask not set
Apr  8 14:44:59 wlceventd: wlceventd_proc_event(491): eth6: Deauth_ind F0:C91:9E:32:31, status: 0, reason: Deauthenticated because sending station is leaving (or has left) IBSS or ESS (3), rssi:0
Apr  8 14:44:59 hostapd: eth6: STA f0:c9:d1:9e:32:31 IEEE 802.11: disassociated
Apr  8 14:45:01 wlceventd: wlceventd_proc_event(527): eth6: Auth F0:C91:9E:32:31, status: Successful (0), rssi:0
Apr  8 14:45:01 wlceventd: wlceventd_proc_event(556): eth6: Assoc F0:C91:9E:32:31, status: Successful (0), rssi:0
Apr  8 14:45:01 hostapd: eth6: STA f0:c9:d1:9e:32:31 IEEE 802.11: associated
Apr  8 14:45:01 kernel: CFG80211-ERROR) wl_cfg80211_change_station : WLC_SCB_AUTHORIZE sta_flags_mask not set
Apr  8 14:45:01 hostapd: eth6: STA f0:c9:d1:9e:32:31 RADIUS: starting accounting session DCE44394FF80B75D
Apr  8 14:45:01 hostapd: eth6: STA f0:c9:d1:9e:32:31 WPA: pairwise key handshake completed (RSN)
Apr  8 14:45:03 wlceventd: wlceventd_proc_event(508): eth6: Disassoc F0:C91:9E:32:31, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Apr  8 14:45:03 wlceventd: wlceventd_proc_event(508): eth6: Disassoc F0:C91:9E:32:31, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Apr  8 14:45:03 hostapd: eth6: STA f0:c9:d1:9e:32:31 IEEE 802.11: disassociated
Apr  8 14:45:03 hostapd: eth6: STA f0:c9:d1:9e:32:31 IEEE 802.11: disassociated
Apr  8 14:45:05 dnsmasq-dhcp[7832]: DHCPDISCOVER(br0) f0:c9:d1:9e:32:31
Apr  8 14:45:05 dnsmasq-dhcp[7832]: DHCPOFFER(br0) 192.168.1.188 f0:c9:d1:9e:32:31
Apr  8 14:45:05 dnsmasq-dhcp[7832]: DHCPDISCOVER(br0) f0:c9:d1:9e:32:31
Apr  8 14:45:05 dnsmasq-dhcp[7832]: DHCPOFFER(br0) 192.168.1.188 f0:c9:d1:9e:32:31
Apr  8 14:45:07 wlceventd: wlceventd_proc_event(527): eth6: Auth F0:C91:9E:32:31, status: Successful (0), rssi:0
Apr  8 14:45:07 wlceventd: wlceventd_proc_event(556): eth6: Assoc F0:C91:9E:32:31, status: Successful (0), rssi:0
Apr  8 14:45:07 hostapd: eth6: STA f0:c9:d1:9e:32:31 IEEE 802.11: associated
Apr  8 14:45:07 kernel: CFG80211-ERROR) wl_cfg80211_change_station : WLC_SCB_AUTHORIZE sta_flags_mask not set
Apr  8 14:45:16 wlceventd: wlceventd_proc_event(491): eth6: Deauth_ind F0:C91:9E:32:31, status: 0, reason: Deauthenticated because sending station is leaving (or has left) IBSS or ESS (3), rssi:0
Apr  8 14:45:16 hostapd: eth6: STA f0:c9:d1:9e:32:31 IEEE 802.11: disassociated
Apr  8 14:45:18 wlceventd: wlceventd_proc_event(527): eth6: Auth F0:C91:9E:32:31, status: Successful (0), rssi:0
Apr  8 14:45:18 wlceventd: wlceventd_proc_event(556): eth6: Assoc F0:C91:9E:32:31, status: Successful (0), rssi:0
Apr  8 14:45:18 hostapd: eth6: STA f0:c9:d1:9e:32:31 IEEE 802.11: associated
Apr  8 14:45:18 kernel: CFG80211-ERROR) wl_cfg80211_change_station : WLC_SCB_AUTHORIZE sta_flags_mask not set
Apr  8 14:45:18 hostapd: eth6: STA f0:c9:d1:9e:32:31 RADIUS: starting accounting session A9FE86951708D6A4
Apr  8 14:45:18 hostapd: eth6: STA f0:c9:d1:9e:32:31 WPA: pairwise key handshake completed (RSN)
Apr  8 14:45:19 dnsmasq-dhcp[7832]: DHCPDISCOVER(br0) f0:c9:d1:9e:32:31
Apr  8 14:45:19 dnsmasq-dhcp[7832]: DHCPOFFER(br0) 192.168.1.188 f0:c9:d1:9e:32:31
Apr  8 14:45:38 wlceventd: wlceventd_proc_event(491): eth6: Deauth_ind 28:6C:07:26:83:2C, status: 0, reason: Unspecified reason (1), rssi:0
Apr  8 14:45:38 wlceventd: wlceventd_proc_event(527): eth6: Auth 28:6C:07:26:83:2C, status: Successful (0), rssi:0
Apr  8 14:45:41 wlceventd: wlceventd_proc_event(491): eth6: Deauth_ind 28:6C:07:26:83:2C, status: 0, reason: Unspecified reason (1), rssi:0
Apr  8 14:45:41 wlceventd: wlceventd_proc_event(527): eth6: Auth 28:6C:07:26:83:2C, status: Successful (0), rssi:0
Apr  8 14:45:43 Mastiff: Got AAE_SIG_REMOTE_CONNECTION_TURNED_ON
Apr  8 14:45:43 rc_service: httpds 12766:notify_rc restart_ddns
Apr  8 14:45:43 start_ddns: update WWW.TUNNELBROKER.NET default@tunnelbroker.net, wan_unit 0
Apr  8 14:45:43 inadyn[9055]: In-a-dyn version 2.8.1 -- Dynamic DNS update client.
Apr  8 14:45:43 inadyn[9055]: Update forced for alias 628579, new IP# 222.**.**.***
Apr  8 14:45:44 inadyn[9055]: Fatal error in DDNS server response:
Apr  8 14:45:44 inadyn[9055]: [401 NI] badauth
Apr  8 14:45:44 inadyn[9055]: Error response from DDNS server, exiting!
Apr  8 14:45:44 inadyn[9055]: Error code 50: Authentication failure
Apr  8 14:45:44 wlceventd: wlceventd_proc_event(491): eth6: Deauth_ind F0:C9:D1:9E:32:31, status: 0, reason: Unspecified reason (1), rssi:0
Apr  8 14:45:44 wlceventd: wlceventd_proc_event(527): eth6: Auth F0:C9:D1:9E:32:31, status: Successful (0), rssi:0
Apr  8 14:45:44 wlceventd: wlceventd_proc_event(556): eth6: Assoc F0:C9:D1:9E:32:31, status: Successful (0), rssi:0
Apr  8 14:45:44 hostapd: eth6: STA f0:c9:d1:9e:32:31 IEEE 802.11: associated
Apr  8 14:45:44 kernel: CFG80211-ERROR) wl_cfg80211_change_station

RMerl, 2021-04-10:
Check your configuration, this is an authentication error.

```
Apr 8 14:45:44 inadyn[9055]: [401 NI] badauth
Apr 8 14:45:44 inadyn[9055]: Error response from DDNS server, exiting!
Apr 8 14:45:44 inadyn[9055]: Error code 50: Authentication failure
```

closed 2021-04-10 by RMerl

suizhenmin, 2021-04-12:
I upgraded from 384 to 386, and before the upgrade, it was normal to set the resolution.