RMerl/asuswrt-merlin.ng #741: RT-AX86U,ipv6 doesn't work with dual-wan

skypjsfly, 2021-04-15:
386.2_2，When I use dual-wan，ipv6 doesn't work.router can't get v6 ip through modem.
If I close dual-wan，It works properly.
And it used to work properly in the past.

nmweizi, 2021-04-17:
me too.
ac86u 386.2_2

jumpsmm7, 2021-05-22:
Welcome to the world of asus incomplete ipv6 implementation. I suspect this issue must be solved by Asus.

JackMerlin, 2021-05-23:
@jumpsmm7 is right. @RMerl can't do anything about it. RMerl can't even debug IPv6 because his ISP does not support IPv6 yet. My suggestion is to use the official firmware to check if it can be reproduced, and then report the this to Asus.

BTW, when you submit a question next time, please use our bug template, because this will help others respond to your question faster.