RMerl/asuswrt-merlin.ng #742: OpenVPN Server on UDP is not working

vipinjn24, 2021-04-18:
**Describe the bug**
OpenVPN Server shows "TLS Handshake failed"

**Model**
RT-AC68U

**Firmware version**
386.2_2

**Client information**
OpenVPN Client 3.2.4

**To Reproduce**
Run the openvpn server on udp.
I have been running on non default port.

**Expected behavior**
It should connect on one go.

**Actual results**

**Will this error be reproduced on the official firmware?**
No, it is working on the official firmware.

**Screenshots**
these are the logs.

Apr 18 19:15:12 ovpn-server1[2383]: PLUGIN AUTH-PAM: initialization succeeded (fg)
Apr 18 19:15:12 ovpn-server1[2383]: PLUGIN_INIT: POST /usr/lib/openvpn-plugin-auth-pam.so '[/usr/lib/openvpn-plugin-auth-pam.so] [openvpn]' intercepted=PLUGIN_AUTH_USER_PASS_VERIFY 
Apr 18 19:15:12 ovpn-server1[2383]: Diffie-Hellman initialized with 2048 bit key
Apr 18 19:15:12 ovpn-server1[2383]: TLS-Auth MTU parms [ L:1621 D:1212 EF:38 EB:0 ET:0 EL:3 ]
Apr 18 19:15:12 ovpn-server1[2383]: TUN/TAP device tun21 opened
Apr 18 19:15:12 ovpn-server1[2383]: TUN/TAP TX queue length set to 1000
Apr 18 19:15:12 ovpn-server1[2383]: do_ifconfig, ipv4=1, ipv6=0
Apr 18 19:15:12 ovpn-server1[2383]: /usr/sbin/ip link set dev tun21 up mtu 1500
Apr 18 19:15:12 ovpn-server1[2383]: /usr/sbin/ip link set dev tun21 up
Apr 18 19:15:12 ovpn-server1[2383]: /usr/sbin/ip addr add dev tun21 10.8.0.1/24
Apr 18 19:15:12 ovpn-server1[2383]: ovpn-up 1 server tun21 1500 1621 10.8.0.1 255.255.255.0 init
Apr 18 19:15:12 ovpn-server1[2383]: Data Channel MTU parms [ L:1621 D:1450 EF:121 EB:406 ET:0 EL:3 ]
Apr 18 19:15:12 ovpn-server1[2383]: Could not determine IPv4/IPv6 protocol. Using AF_INET6
Apr 18 19:15:12 ovpn-server1[2383]: Socket Buffers: R=[122880->122880] S=[122880->122880]
Apr 18 19:15:12 ovpn-server1[2383]: setsockopt(IPV6_V6ONLY=0)
Apr 18 19:15:12 ovpn-server1[2383]: UDPv6 link local (bound): [AF_INET6][undef]:21194
Apr 18 19:15:12 ovpn-server1[2383]: UDPv6 link remote: [AF_UNSPEC]
Apr 18 19:15:12 ovpn-server1[2383]: MULTI: multi_init called, r=256 v=256
Apr 18 19:15:12 ovpn-server1[2383]: IFCONFIG POOL IPv4: base=10.8.0.2 size=252
Apr 18 19:15:12 ovpn-server1[2383]: Initialization Sequence Completed
Apr 18 19:15:28 ovpn-server1[2383]: MULTI: multi_create_instance called
Apr 18 19:15:28 ovpn-server1[2383]: 10.1.168.6:51770 Re-using SSL/TLS context
Apr 18 19:15:28 ovpn-server1[2383]: 10.1.168.6:51770 Control Channel MTU parms [ L:1621 D:1212 EF:38 EB:0 ET:0 EL:3 ]
Apr 18 19:15:28 ovpn-server1[2383]: 10.1.168.6:51770 Data Channel MTU parms [ L:1621 D:1450 EF:121 EB:406 ET:0 EL:3 ]
Apr 18 19:15:28 ovpn-server1[2383]: 10.1.168.6:51770 Local Options String (VER=V4): 'V4,dev-type tun,link-mtu 1541,tun-mtu 1500,proto UDPv4,auth SHA1,keysize 128,key-method 2,tls-server'
Apr 18 19:15:28 ovpn-server1[2383]: 10.1.168.6:51770 Expected Remote Options String (VER=V4): 'V4,dev-type tun,link-mtu 1541,tun-mtu 1500,proto UDPv4,auth SHA1,keysize 128,key-method 2,tls-client'
Apr 18 19:15:28 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:51770: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:15:28 ovpn-server1[2383]: 10.1.168.6:51770 TLS: Initial packet from [AF_INET6]::ffff:10.1.168.6:51770, sid=be496f45 f4d507d8
Apr 18 19:15:28 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 WRITE [26] to [AF_INET6]::ffff:10.1.168.6:51770: P_CONTROL_HARD_RESET_SERVER_V2 kid=0 [ 0 ] pid=0 DATA len=0
Apr 18 19:15:29 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:51770: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:15:29 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 WRITE [22] to [AF_INET6]::ffff:10.1.168.6:51770: P_ACK_V1 kid=0 [ 0 ]
Apr 18 19:15:30 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:51770: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:15:30 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 WRITE [26] to [AF_INET6]::ffff:10.1.168.6:51770: P_CONTROL_HARD_RESET_SERVER_V2 kid=0 [ 0 ] pid=0 DATA len=0
Apr 18 19:15:31 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:51770: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:15:31 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 WRITE [22] to [AF_INET6]::ffff:10.1.168.6:51770: P_ACK_V1 kid=0 [ 0 ]
Apr 18 19:15:32 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:51770: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:15:32 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 WRITE [22] to [AF_INET6]::ffff:10.1.168.6:51770: P_ACK_V1 kid=0 [ 0 ]
Apr 18 19:15:33 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:51770: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:15:33 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 WRITE [22] to [AF_INET6]::ffff:10.1.168.6:51770: P_ACK_V1 kid=0 [ 0 ]
Apr 18 19:15:34 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:51770: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:15:34 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 WRITE [26] to [AF_INET6]::ffff:10.1.168.6:51770: P_CONTROL_HARD_RESET_SERVER_V2 kid=0 [ 0 ] pid=0 DATA len=0
Apr 18 19:15:35 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:51770: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:15:35 ovpn-server1[2383]: 10.1.168.6:51770 UDPv6 WRITE [22] to [AF_INET6]::ffff:10.1.168.6:51770: P_ACK_V1 kid=0 [ 0 ]
[... 174 lines cut ...]
Apr 18 19:16:55 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 WRITE [26] to [AF_INET6]::ffff:10.1.168.6:37420: P_CONTROL_HARD_RESET_SERVER_V2 kid=0 [ 0 ] pid=0 DATA len=0
Apr 18 19:16:56 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:37420: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:16:56 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 WRITE [22] to [AF_INET6]::ffff:10.1.168.6:37420: P_ACK_V1 kid=0 [ 0 ]
Apr 18 19:16:57 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:37420: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:16:57 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 WRITE [22] to [AF_INET6]::ffff:10.1.168.6:37420: P_ACK_V1 kid=0 [ 0 ]
Apr 18 19:16:58 ovpn-server1[2383]: 10.1.168.6:53309 TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)
Apr 18 19:16:58 ovpn-server1[2383]: 10.1.168.6:53309 TLS Error: TLS handshake failed
Apr 18 19:16:58 ovpn-server1[2383]: 10.1.168.6:53309 SIGUSR1[soft,tls-error] received, client-instance restarting
Apr 18 19:16:58 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:37420: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:16:58 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 WRITE [22] to [AF_INET6]::ffff:10.1.168.6:37420: P_ACK_V1 kid=0 [ 0 ]
Apr 18 19:16:59 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:37420: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:16:59 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 WRITE [26] to [AF_INET6]::ffff:10.1.168.6:37420: P_CONTROL_HARD_RESET_SERVER_V2 kid=0 [ 0 ] pid=0 DATA len=0
Apr 18 19:17:00 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 READ [14] from [AF_INET6]::ffff:10.1.168.6:37420: P_CONTROL_HARD_RESET_CLIENT_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:17:00 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 WRITE [22] to [AF_INET6]::ffff:10.1.168.6:37420: P_ACK_V1 kid=0 [ 0 ]
Apr 18 19:17:07 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 WRITE [14] to [AF_INET6]::ffff:10.1.168.6:37420: P_CONTROL_HARD_RESET_SERVER_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:17:08 ovpn-server1[2383]: 10.1.168.6:37443 TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)
Apr 18 19:17:08 ovpn-server1[2383]: 10.1.168.6:37443 TLS Error: TLS handshake failed
Apr 18 19:17:08 ovpn-server1[2383]: 10.1.168.6:37443 SIGUSR1[soft,tls-error] received, client-instance restarting
Apr 18 19:17:18 ovpn-server1[2383]: 10.1.168.6:50584 TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)
Apr 18 19:17:18 ovpn-server1[2383]: 10.1.168.6:50584 TLS Error: TLS handshake failed
Apr 18 19:17:18 ovpn-server1[2383]: 10.1.168.6:50584 SIGUSR1[soft,tls-error] received, client-instance restarting
Apr 18 19:17:23 ovpn-server1[2383]: 10.1.168.6:37420 UDPv6 WRITE [14] to [AF_INET6]::ffff:10.1.168.6:37420: P_CONTROL_HARD_RESET_SERVER_V2 kid=0 [ ] pid=0 DATA len=0
Apr 18 19:17:53 ovpn-server1[2383]: 10.1.168.6:37420 TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)
Apr 18 19:17:53 ovpn-server1[2383]: 10.1.168.6:37420 TLS Error: TLS handshake failed
Apr 18 19:17:53 ovpn-server1[2383]: 10.1.168.6:37420 SIGUSR1[soft,tls-error] received, client-instance restarting

vipinjn24, 2021-04-18:
TCP is working fine.

RMerl, 2021-04-18:
Works for me.  Check your configuration, especially your WAN MTU - an invalid MTU will prevent UDP from properly negotiating.

vipinjn24, 2021-04-18:
So how exactly to get the mtu for my connection?
Do i need to contact my isp?

RMerl, 2021-04-18:
Go to the WAN page, and set it to 1500.

vipinjn24, 2021-04-18:
It didnt work either.

vipinjn24, 2021-05-09:
I did some more research.

ping -f address -l 1500
this command shows "Packet needs to be fragmented but DF set."

The maximum i can reach for my router either public ip or internal ip is 1472.

Please help me out with resolving this issue.

my client config file:
client
dev tun
proto udp
remote XXX.XXX.XXX.XXX 4911
resolv-retry infinite
nobind
float
ncp-ciphers CHACHA20-POLY1305:AES-128-GCM:AES-256-GCM:AES-128-CBC:AES-256-CBC
keepalive 15 60
auth-user-pass
remote-cert-tls server
//**Removed the keys**//

**Server Config**
[image]

RMerl, 2021-05-09:
The issue tracker is only for development purposes.  If you need technical support or have a question, please use the support forums at https://www.snbforums.com/forums/asuswrt-official.51/ instead.

vipinjn24, 2021-05-10:
I dont know how this is possible that the server is working with same exact configuration when i was using official firmware but now its not working any more.

I believe this is the issue with your builds.
Please check your builds again while i am going to post this question on the support forums.

vipinjn24, 2021-05-10:
Ok got it from support...
Merlin firmware do have NAT hair-pinning that is why it is not working for devices within the same network.
However connecting from internet was good.