RMerl/asuswrt-merlin.ng #743: Traffic Analyzer Daily and Monthly Statics are broken

FastThenLeft, 2021-04-18:
**Describe the bug**

The Traffic Analyzer > Traffic Monitor > Daily and Monthly Statics are broken in 386.2_2 in RT-AC5300. They worked fine in 386.1_2.

**Model**

RT-AC5300

**Firmware version**

386.2_2
NOTE: I ran Administration > Factory default > Restore (with the box check to initialize all settings) and still have the problem.

**Client information**

Chrome 90.0.4430.72 & Firefox 87.0.

**To Reproduce**

Download 3-4 GB of data.
Look at Traffic Analyzer > Traffic Monitor > Daily and it will show:
Reception: 17,179,869,182.80 GB
Transmission: 17,179,869,181.43 GB
Total: 34,359,738,364.24 GB

Look at Traffic Analyzer > Traffic Monitor > Monthly and it will show:
No data in table.

**Expected behavior**

I would expect Daily and Total to show the 3-4 GB I downloaded that day.
NOTE: 386.1_2 worked fine.

**Actual results**

Download 3-4 GB of data.
Look at Traffic Analyzer > Traffic Monitor > Daily and it will show:
Reception: 17,179,869,182.80 GB
Transmission: 17,179,869,181.43 GB
Total: 34,359,738,364.24 GB

Look at Traffic Analyzer > Traffic Monitor > Monthly and it will show:
No data in table.

**Will this error be reproduced on the official firmware?**

**Screenshots**

[image]

**Additional context**

This has worked on all versions up to 386.1_2. After installing 386.2_2 I did a factory reset.

RMerl, 2021-04-18:
Works for me.  Make sure you did create the database under Tools -> Other Settings, and look in your System Log for rstats related errors when accessing the database.

FastThenLeft, 2021-04-18:
Under Tools > Other Settings > Traffic history location it's set to "RAM
(Default)".

RMerl, 2021-04-18:
RAM means the database will get wiped every reboot.  You need to create one on a USB stick.

FastThenLeft, 2021-04-18:
Yes, I know that. But to debug this I put it in RAM to eliminate as many
variables as possible. I have the same problem in RAM or on USB. After
about 2 hours it shows that I've downloaded 17,000,000,000 GB! I've done a
factory reboot and a power cycle. Same problem.

wolffcatsky, 2021-04-18:
> Yes, I know that. But to debug this I put it in RAM to eliminate as many variables as possible. I have the same problem in RAM or on USB. After about 2 hours it shows that I've downloaded 17,000,000,000 GB! I've done a factory reboot and a power cycle. Same problem.
> […](#)

I have the same very annoying problem.

RMerl, 2021-04-18:
> have the same very annoying problem, he's said before it's an Asus issue. While I understand that, I wish Merlin would use his super powers (his contacts at Asus) to help them fix ths.

Already did that...

https://www.snbforums.com/threads/traffic-spike-fix-from-384-14_2-was-reverted-in-386-1.69843/post-678809

Asus fixed that issue on newer HND models weeks ago, however neither of us have been able to reproduce the problem on older models.  I recommend you install the stock firmware, reproduce the issue there, then contact them through the Feedback form as I indicated in that post.